luc-github/ESP3D #372: Connecting ESP8266 to a Bigtreetech SKR 1.3

kkberg, 2019-09-19:
I'm just trying to connect the ESP8266 to my SKR 1.3 board with a stepdown from 5v to 3.3v.
I connected Rx and Tx also converted power 3.3 from SKR 1.3. . I'm an electronics NOOB and tried my best to get it working. But only connecting 5v GND Rx Tx does not start Wifi on the ESP. I can't see the Wifi in that way. Connecting ESP to the programmer again, Wifi appears without any problem. SKR 1.3 has an AUX connector with 2x4 pins and one in parallel with 5 pins, reset pin included. 

I found a schematics here http://stefanfrings.de/esp8266/ with pullups and a capacitor. 
Is it really needed to get it running? 

has someone else same problem and a solution.

luc-github, 2019-09-19:
What board of esp8266 are you using?
Do you see esp3d Access point is only connect GND and +5V?

kkberg, 2019-09-19:
I'm using this one https://www.az-delivery.de/products/esp8266-01s-mit-usb-adapter?_pos=15&_sid=4cd6e8f08&_ss=r&ls=de
It's an ESP8862 ESP-01S

Can't see access point when wiring GND and 3.3v.

luc-github, 2019-09-19:
This board is esp01 check wiki need all pins connected

luc-github, 2019-09-20:
@kkberg is that ok ?

wilbur4321, 2019-09-25:
According to the schematics, the skr v1.3 has 3.3v available on pin2 of the ICSP (J2) connector.  I suspect, but cannot say for sure, that the IO signals are all 3.3v rather than 5v.
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-SCH.pdf

sl1pkn07, 2019-09-27:
i think you can use this in the tft port (if you configure the pins correctly(?))

https://www.keyestudio.com/keyestudio-esp-01s-wifi-to-serial-shield-module-for-arduino-esp8266-wifi-p0499-p0499.html

weed2all, 2019-09-30:
I use nodemcu v3 and an esp 8266-01s with my skr 1.1 and skr 1.3 and fully work...on my nodemcu I use the esp3d v2.1 and on my esp8266-01s I use esp3d v1.1
For stepdown the voltage from 5v to 3.3v I use this buck converter
 € 0,81 | Alta Qualidade 5 PCS 5 V a 3.3 V Para Buck Módulo DC-DC Step-Down fonte de Alimentação LDO AMS1117 800MA
https://s.click.aliexpress.com/e/3mc7zA6G

kkberg, 2019-09-30:
> 
> 
> Any update ?

Sorry for my late response, ordered a special adapter https://www.amazon.de/gp/product/B07DXPXF1Y/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1 .
This should help to connect only 4 wires to the TFT port. I'll keep you informed when having tested. But will take a while because of shipping date.

weed2all, 2019-09-30:
I made an adapter for my esp8266-01s witch contains on 2 × Dupont 4pin shells that are glued togheter  back to back, and that little buck converter witch is glued with kapton tape to the 4 pin Dupont shell...in order to apply 3.3v to esp...from skr1.3 AUX-1 header came 4 wires...red is 5v witch go to v-in on buck converter, black is grount witch go to buck converter and to esp gnd pin, green witch is TX from skr1.3 board go to RX on esp, and blue witch is RX from skr 1.3 board witch go to TX on esp, and another small red wire witch go from v-out of buck converter to the esp 3.3v pin

kkberg, 2019-09-30:
> I made an adapter for my esp8266-01s witch contains on 2 × Dupont 4pin shells that are glued togheter back to back, and that little buck converter witch is glued with kapton tape to the 4 pin Dupont [...]

Okay, that is what the adapter from Amazon is also doing for 2,10€ too. Tinkering your way mostely end in smoke when I'm doing such things. :)

weed2all, 2019-09-30:
It is easy....only need to connect the right wire to the right pin...and I didnt was in mood to wait 4 weeks to arrive that adapter...

weed2all, 2019-09-30:
But you made a smart decision to buy the adapter!

kkberg, 2019-09-30:
> 
> 
> It is easy....only need to connect the right wire to the right pin...and I didnt was in mood to wait 4 weeks to arrive that adapter...

Sadely this isn't working for me, because wifi doesn't start and that is the reason of my question here.

weed2all, 2019-09-30:
Where do you connect the esp on skr board?did you setup the 2nd serial port on marlin?

weed2all, 2019-09-30:
If you had not set the 2nd serial port on marlin will not work...

kkberg, 2019-09-30:
> 
> 
> Where do you connect the esp on skr board?did you setup the 2nd serial port on marlin?

I connected it to the AUX1 TFT 5pin connector.

kkberg, 2019-09-30:
> 
> 
> If you had not set the 2nd serial port on marlin will not work...

This is secondary, because more important is the not working wifi on ESP, when connected to serial port. But second serial is rnablrd in Marlin.

weed2all, 2019-09-30:
I use esp8266-01s, I think you have esp8266-01
Those are not working...

kkberg, 2019-09-30:
> 
> 
> I use esp8266-01s, I think you have esp8266-01
> Those are not working...

I only have esp8266-01S !!!!

weed2all, 2019-09-30:
On the image witch one do you have? The left or the right one?
[image]

weed2all, 2019-09-30:
That is the same I have...and it works very well

kkberg, 2019-09-30:
> 
> 
> On the image witch one do you have? The left or the right one?
> [image]

I have the right one named as ESP-01s

luc-github, 2019-09-30:
be sure when you flash you use erase all flash seting

kkberg, 2019-09-30:
> 
> 
> be sure when you flash you use erase all flash setting

I used fresh ESP for flashing, directly out of the box. I also tried to flash blank.bin first

luc-github, 2019-09-30:
and even esp8266 sample code does not start ?

kkberg, 2019-09-30:
> 
> 
> and even esp8266 sample code does not start ?

Plug into the programmer everything works fine. ESP3D works and any other sketch too. Only whend hooked up to the board wifi doesn't start.

luc-github, 2019-09-30:
if work on programmer it is working 
it means your printer wiring is incorrect 
it is an esp 01 so it means all pins must be connected to something tx/ rx/ gnd and all others to 3.3v  ( vcc/ en/ gpio0/ etc)

luc-github, 2019-09-30:
missing connection will result esp01 not starting properly in user mode

weed2all, 2019-09-30:
Use nodemcu flasher with blank 1mb sketch to completely erase the esp and afterward install again esp3d...should work...I confirm mine work very well...

kkberg, 2019-09-30:
https://www.youtube.com/watch?v=G6zKZyWLTDU
got it to work, don't know what it was. Thank you all for your help!!!!!

closed 2019-09-30 by kkberg

kkberg, 2019-09-30:
After being back home I examined my 6 ESP's. I ordered all as ESP8266 01-S, but they sent 4 different ones, which are not labeled but looking nearly the same. So I now have 2 "real" ESP8266 01-S and 4 seem to be "only" ESP8266.  I was so concentrated on my project and not on the parts itself. So I apologize for making that mistake and giving some wrong information. I hope that the ordered adapter will fix the problem for the only ESP8266.

Thank you all

Klaus

weed2all, 2019-09-30:
Esp8266 will not work...i have a bunch of them and they dont work...only esp8267-01s

kkberg, 2019-09-30:
I could not believe that I got different types, because I ordered only 01S. But important that I found it because of your help here. Thank you again. I'm now figuring out to find a sketch which only provides a serial port over wifi to connect the printer via wifi to my Repetier server.

sl1pkn07, 2019-09-30:
the keyestudio unit works with normal 8266 (non 01S)

luc-github, 2019-09-30:
esp8266 have only 512Kb flash when 01-S have 1MB flash
you may have the 512Kb working by disabling most of feature and just keep
````
//SERIAL_COMMAND_FEATURE: allow to send command by serial
#define SERIAL_COMMAND_FEATURE

//TCP_IP_DATA_FEATURE: allow to connect serial from TCP/IP
#define TCP_IP_DATA_FEATURE

//WS_DATA_FEATURE: allow to connect serial from Websocket
#define WS_DATA_FEATURE
```
⚠️ and follow this : https://github.com/luc-github/ESP3D/issues/331 but you won't have webupdate available neither on 512Kb

which on my side gave me `Sketch uses 374150 bytes `

for webUI : this one has only english 
[index.html.gz](https://github.com/luc-github/ESP3D/files/3671427/index.html.gz)

and this one is ultra gziped
[index.html.gz](https://github.com/luc-github/ESP3D/files/3671430/index.html.gz)

so you should be able to use 128KB for SPIFFS on 512KB flash

luc-github, 2019-09-30:
@sl1pkn07 you have `Chip is ESP8266EX` not `ESP8266` - actually I though ESP8266 with 512KB were no more sold and even available

sl1pkn07, 2019-09-30:
my unit is sold as non S, and have the non S layout

<details>
[image]
</details>

luc-github, 2019-09-30:
wow my 512KB branch is now 4  years old!!! and the files fitted the 64KB partition ^_^
https://github.com/luc-github/ESP3D/tree/ESP-512K-64KSPIFFS

weed2all, 2019-09-30:
Think that some chinese guys still sell those esp8266...i have aswell esp8266 n on s with 1mb...they arent work for me with esp3d...i install esp3d on my esp8266-01s and worked first time

luc-github, 2019-09-30:
@sl1pkn07 yes ESP8266EX came after ESP8266 , they were the first with 1MB as 512KB was already really tight

weed2all, 2019-09-30:
Exactly

sl1pkn07, 2019-09-30:
for me works. but i cant self-updating via web

my changes

~~~
└───╼  git diff HEAD^
diff --git a/esp3d/config.h b/esp3d/config.h
index 37e7fc3..34da355 100644
--- a/esp3d/config.h
+++ b/esp3d/config.h
@@ -94,7 +94,7 @@
 //Extra features /////////////////////////////////////////////////////////////////////////
 
 //Workaround for Marlin 2.X coldstart
-//#define DISABLE_CONNECTING_MSG
+#define DISABLE_CONNECTING_MSG
 
 //Serial rx buffer size is 256 but can be extended
 #define SERIAL_RX_BUFFER_SIZE 512
diff --git a/platformio.ini b/platformio.ini
index e713cba..72ed7eb 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -14,6 +14,7 @@ build_dir   = .pioenvs
 lib_dir     = libraries
 libdeps_dir = .piolibdeps
 data_dir = esp3d/data
+default_envs = esp8266
 
 [env:esp32dev]
 platform = https://github.com/platformio/platform-espressif32.git
@@ -35,7 +36,7 @@ lib_ignore =
 
 [env:esp8266]
 platform = espressif8266
-board = esp12e
+board = esp01_1m
 framework = arduino
 monitor_speed = 115200
 ; set frequency to 160MHz
@@ -45,7 +46,7 @@ board_build.f_flash = 40000000L
 board_build.flash_mode = dio
 upload_resetmethod = ck
 build_flags = 
-    -Wl,-Teagle.flash.4m2m.ld
+    -Wl,-Teagle.flash.1m512.ld
     -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
     -DNONOSDK221=1
     -DNDEBUG
~~~

luc-github, 2019-09-30:
yes with 1M flash it should always works but webUpdate may not be working depending of FW size

luc-github, 2019-09-30:
also as SPIFFS is now a critical size vs webUI it is important to format and upload just after the webUI file
deleting webui when another file is also present (like preferences.json or any other file) will limit the size available even it says it is enough - the available size will be the one from the previous deleted file

weed2all, 2019-09-30:
Luc...offtopic question...does esp3d work with reprap firmware?I have a radds 1.6 and arduino due and I install reprap firmware on it...

sl1pkn07, 2019-09-30:
yep. my compiled firm has 489KB , my total space (after burned it) 471.36 KB and 124.27 KB used by preferences.ini and index.html.gz

luc-github, 2019-09-30:
@weed2all yes if you enable Marlin compatibility when you flash reprap

weed2all, 2019-09-30:
Sweet...I must give it a try!

kkberg, 2019-10-04:
With this adapter the "smaller" ESP8266 works fine with ESP3D.

[image]
[image]
[image]

ducker, 2019-10-16:
Hi, I have some troubles with connecting ESP to SKR 1.3. I connected it to TFT port and I use the same adapter as mentioned above. I also tried to connect ESP directly to SKR pins without adapter. ESP3D is working fine but I don't receive any message from the board. I can send some commands like set temp but nothing appears on the gcode console or sd card management window. When I loop rx and tx on ESP I see the echo so it looks for me like board don't send the responses. Have you experienced similar behaviour?

kkberg, 2019-10-16:
Hi, you have to enable sevond serial port in Marlin. I have same config with skr 1.3 and it's working fine. I am not at home in the and back this evening.

ducker, 2019-10-17:
Yes, it is crossed. If not it will not work at all and I can send commands successfully. Strange behaviour. I think that SKR board may be broken, but I have no idea how to easily confirm it.

hapklaar, 2019-11-17:
> Yes, it is crossed. If not it will not work at all and I can send commands successfully. Strange behaviour. I think that SKR board may be broken, but I have no idea how to easily confirm it.

@ducker I think I'm having the same issue. I can send GCODE ESP>SKR, but am not getting anything in return. Also M117 output from board is not reaching anything connected through TFT/ESP.

Were you able to get this fixed?