luc-github/ESP3D #373: Click for resume....

CsepregiArtur, 2019-09-19:
Hello

When I start my printer without the D1 mini everything is oke working fine.
My Motherboard is a Bigtreetech skr v1.3, with Marlin 2.0 and I'm using a D1 Mini for esp3d .
When I start it whit the D1 mini I receive the message Click to resume, and I need to "Click " for the ESP3D to connect.

Do you Know  a solution for it?

kkberg, 2019-09-19:
Hello,
I'm just tinkering around wit an ESP8266 and SKR 1.3 and a bit confused how to get it working. I connected Rx Tx Vcc GND with a stepdown to 3.3v. ESP in't starting Wifi. How did you manage it with the D1 mini and which of the AUX did you use?

Thank you

luc-github, 2019-09-19:
Where message come from?    
How did you wired your board ?

kkberg, 2019-09-19:
Whom do you mean?

CsepregiArtur, 2019-09-20:
@kkberg I used the aux1 port, in the ESP3D wiki you have a picture how to make conection
https://github.com/luc-github/ESP3D/wiki/D1-mini
 to a D1 mini, i didn't use any step down, 5v from the Aux pin, description yo have here https://github.com/luc-github/ESP3D/wiki/MKS-GEN-v1.2-(1.3-and-above-maybe) but it is simple 5v GND TX RX.

CsepregiArtur, 2019-09-20:
@luc-github 

Where message come from? - on the lcd screen (1284 lcd smart controller). when i start the printer with the D1 mini conected apers a message Click to resume... and i need to push the knob to continue and after thar the Esp3d comes to live :)
How did you wired your board ? I used the Aux 1 acording to this description. https://github.com/luc-github/ESP3D/wiki/MKS-GEN-v1.2-(1.3-and-above-maybe)
firt I thinkt it is oke and it needs to work so, but I saw this video from Chris Riley
https://www.youtube.com/watch?v=br7BryjWPJU
oke he is using Panucatt Wifi backpack but it is using outher pins pins abouve Aux-1 labeled TFT 
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR%20V1.3%20Wiring%20diagram.jpg
and when he is povering on the printer no message apears, only the IP address, but by me is the Click to resume message.

luc-github, 2019-09-20:
OK then message come from Marlin - need to check Marlin FW why it need confirmation to connect, should be some Marlin settings, I have used offcial Marlin FW 2.0 without such message, so should be something specific to BigTreeTech code - better to debug or ask here : https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3 why this question is raised when connect serial. 
I do not have 1.3  so  cannot help too much on it sorry

CsepregiArtur, 2019-09-20:
@luc-github my problem is that I asked alredy from bigtreetech.
than I think i will compare your marlin version from github with my version, hope to find the problem :)
after that if you don't mind can i create a page on wiki for those who have SKR v1.3

Thank you for your help

luc-github, 2019-09-20:
@CsepregiArtur where did you asked ? I do not see such question on issue tracker

CsepregiArtur, 2019-09-20:
I asked them o client support  from aliexpress :(
but i posted also the question on github right now.

luc-github, 2019-09-20:
@CsepregiArtur did you flashed your FW by your self ?
is EXTENSIBLE_UI enabled in your configuration.h ?

if not message is also displayed due to GCODE command : https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/firmware/Marlin-bugfix-2.0.x/Marlin/src/gcode/lcd/M0_M1.cpp#L75
if there is M1 or M0 sent to printer -  could be a good start to debug 
To disable it : disable HAS_RESUME_CONTINUE in /src/inc/Conditionals_LCD.h

CsepregiArtur, 2019-09-20:
@luc-github Yes I did flash my self, i download it from marlins web page.
EXTENSIBLE_UI  is not enabled i will enabel it as soon as I arive home, need to test it, and alos i will disabel the HAS_RESUME_CONTINUE, after that I will com back with update. thank you for your help!!!

luc-github, 2019-09-20:
If EXTENSIBLE_UI is not enabled. Is good. Do not enable it

CsepregiArtur, 2019-09-20:
Hello

After disabling i can't comply  HAS_RESUME_CONTINUE in /src/inc/Conditionals_LCD.h

and i have this message on esp3d echo:busy: paused for user

luc-github, 2019-09-20:
Was blind try as I do have the board
Need to find out why you M0 or M1 command. Is also happen with only RX or only TX connectes ?

CsepregiArtur, 2019-09-21:
yes, only happens if the RX TX is connected.
and I think I found something: when I start the printer and wait to boot up until it appears the screen with the info, temperature and so on, and power up the D1 mini then, I don't get the error message. :)
Now only I need to figure up how to delay the power or the start of the D1 mini.
Do you have a solution where I can add a delay in the start of the program?
or I need to get power from somewhere else...

luc-github, 2019-09-21:
It is already the case, there are 8000ms delay by default
My question was RX only = RX connected and TX not connected. TX only = RX not connected and TX connected

CsepregiArtur, 2019-09-22:
hello

with RX disconnected nothing happens, With TX I receive message connecting, and that is al.
Can I change that default 8000ms to something else?
I would like to increase it.

luc-github, 2019-09-22:
RX from printer ? or RX from ESP3D ?

here https://github.com/luc-github/ESP3D/blob/2.1/esp3d/esp3d.ino#L29
`myesp3d.begin(15000);` will delay 15s

CsepregiArtur, 2019-09-22:
I unplugged the wiers from my printer.

luc-github, 2019-09-22:
yes I guess 😸:  not connected = unplugged

1 - RX wemos D1 go to TX printer
2 - TX  wemos D1 go to RX printer

so which connection generate the message ?

CsepregiArtur, 2019-09-22:
sorry for that

with RX from printer TX D1 disconnected nothing happens, With TX from printer RX D1 I receive message connecting, and that is al.
so the second option from you message gives me the connecting message

luc-github, 2019-09-22:
Ok so ESP send M0 or M1 to printer somehow. may be the initial garbage message, 
to check I suggest to power esp from another source and and power esp first then connect to printer then power printer - if it is garbage output you should not have issue

CsepregiArtur, 2019-09-22:
Meanwhile I found in marlin this section

#define TX_BUFFER_SIZE 0
//#define RX_BUFFER_SIZE 1024
I read somewhere it could be buffer overflow problem too.
So I will give a try of changing those values

CsepregiArtur, 2019-09-22:
Hello

I have the solution //#define RX_BUFFER_SIZE 1024 needs to be enabled, the error is caused by Buffer overflow I think :) but after I defined RX_BUFFER_SIZE 1024 it connects at is sold without any message 

@luc-github please close the question :) and may thaks for your help!

luc-github, 2019-09-22:
good - have fun

closed 2019-09-22 by luc-github

PaschyS, 2020-01-11:
This solution will not work with the skr pro and the tft35v3. What can I do for help? ESP is connected at the tft.