luc-github/ESP3D #374: [QUESTION] How do Marlin Wifi settings have influence to ESP3D

kkberg, 2019-09-20:
Good morning,

I can't get information about the Marlin Wifi settings and what realley happens when setting them.
What modul is needed to use that settings? Maybe someone has detailed information for sharing.

Thank you very much

luc-github, 2019-09-20:
What Marlin WiFi settings are you talking about ?

kkberg, 2019-09-20:
I mean this out of configuration_adv.h

/**
 * WiFi Support (Espressif ESP32 WiFi)
 */
//#define WIFISUPPORT
#if ENABLED(WIFISUPPORT)
  #define WIFI_SSID "Wifi SSID"
  #define WIFI_PWD  "Wifi Password"
  //#define WEBSUPPORT        // Start a webserver with auto-discovery
  //#define OTASUPPORT        // Support over-the-air firmware updates
#endif

luc-github, 2019-09-20:
they are for Marlin installed on ESP32 as printer board they are not for ESP3D but Marlin FW - they are not used for a daugther board like you do in your previous open question

kkberg, 2019-09-20:
Thank you, that makes sence. I'm confused with all that stuff, need weekend. :)

kkberg, 2019-09-20:
clarified, thank you

kkberg, 2019-09-20:
.. you sea:)

closed 2019-09-20 by kkberg

luc-github, 2019-09-20:
NP
FYI : there is a version of Marlin that support ESP3D for esp32 https://github.com/luc-github/Marlin