luc-github/ESP3D #377: [BUG]

emsoft1, 2019-09-26:
**Describe the bug**
hi guys i have anet a8 and i try to use esp3d and after many try  finally i install the 1.1 version (i have esp8266 1m flash version).

**To Reproduce**
Steps to reproduce the behavior:
after setting up i try to move the 3d printer  in different axis and check the temp everything is working 
but when i want to send the gcode to the 3d printer i get "upload fail error " in my 3dprinter 
i try different firmware for marlin ,....but it is not working every time i get same error 
in some of the frimware selesction even i can not see the sd card content and even i can not order to printer from existing gcode in the sd card in any of the version that i select in esp3d 
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:**
- ESP core version: 1.1
- Version:[e.g 2.1b4]
- Web UI Version:1.1 
- Wifi mode:esp8266 1mb flash 
- WebUI version: [e.g. 2.0b4]
- Any change done in FW  

**Target Firmware:**
 - Name: [anet a8 
 - Version [e.g. 2.0]

**Board used (please complete the following information):**
 - MCU: [e.g. ESP8266] 
 - Name:esp8266
 - Flash size: 1mb 

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

luc-github, 2019-09-26:
for upload and marlin be sure you use latest esp3d 2.1 and latest webui 2.1, git one and esp8266 2.5.2 core

luc-github, 2019-09-26:
what version of marlin do you have?

luc-github, 2019-09-26:
be sure you can see sd content from webui and from serial using M20

closed 2019-10-26 by stale[bot]