luc-github/ESP3D #379: [BUG] Cannot start afte

huczas, 2019-09-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- WebUI version: [e.g. 2.0b4]
- Any change done in FW  

**Target Firmware:**
 - Name: [e.g. Marlin]
 - Version [e.g. 2.0]

**Board used (please complete the following information):**
 - MCU: [e.g. ESP32] 
 - Name:[e.g. NodeMCU2s] 
 - Flash size: [e.g. 4M: 2M/2M]

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

luc-github, 2019-09-30:
yes web UI and FW start to grow with the time  - I think I need to do mono language version of webUI to decrease size

what space did you tried that is not working ?

huczas, 2019-09-30:
Any above SPIFFS 128 didn’t start. 
Premade UI with proper languages are good choice 👍 nice to try that approach.

luc-github, 2019-09-30:
this is weird people having 1M flash where using 194kb SPIFFS 
but you may need to use the erase all flash setting

huczas, 2019-09-30:
Ok, I will try that at the evening. I’m also afraid that I have something wrong with libraries that I have. 
Is there any chance that you can build binary file for me?

luc-github, 2019-09-30:
you cannot be wrong if you use the libraries present in repository libraries directory. need to copy them as explained in read me to be sure you use the one I tested and certified working

huczas, 2019-10-02:
Ok, so I've ended with 1M (160k SPIFFS) - that I can confirm is working with esp8266 library v2.5.2.
I've disabled notifications to get little more space.
```
//NOTIFICATION_FEATURE : allow to push notifications
//#define NOTIFICATION_FEATURE
```

New question is: if in the future I want to upgrade firmware using web - how to prepare file? or are you going to prepare next version ready to upload?

luc-github, 2019-10-02:
in arduino just generate  binary, it will be in sketch folder
[image]

Due to all possible configurations I cannot build all binary

huczas, 2019-10-02:
ok, so there is not any specific naming of that binary - all is clear to me now. Thanks for your work! I will buy you a coffee 😉

closed 2019-10-02 by huczas