luc-github/ESP3D #380: [BUG]

kr15uk, 2019-10-03:
**Describe the bug**
Can't upload index.html.gz
"Update failed(3): File write failed" pop up

**To Reproduce**
Steps to reproduce the behavior:
After FW flash go to the web interface, choose files, upload.

**Expected behavior**
Expect WEBUI to be uploaded and get button go to WEBUI (like it happens on NodeMCU's of mine)

**Screenshots**

**ESP3D Firmware:**
- ESP core version: 2.5.2
- Version: v2.1.0.b39
- Web UI Version: 2.1
- Wifi mode: STA 
- WebUI version: 2.1
- Any change done in FW NetBIOS disabled

**Target Firmware:**
 - Name: Marlin
 - Version 2

**Board used (please complete the following information):**
 - MCU: ESP8826
 - Name: ESP-01 
 - Flash size: 1M (512K SPIFFS)

**Additional context**
1. Flashed with 512K SPIFFS, if using recommended 128K SPIFFS as per Wiki I get not enough space hence due to the WEBUI 2.1 index.html.gz file now being 127K (shows 110K available).
2. Also, have done esptool erase flash prior flashing FW and [ESP710]FORMAT via serial monitor prior upload.

luc-github, 2019-10-03:
yes webui is now a bottle neck for 1M Flash. I will work on a splited language version to decrease the size. as now 128k is too tight

kr15uk, 2019-10-03:
Sorry, looks like Additional content "took over" the ticket issue.

Just to clarify:
1M (128K SPIFFS) - not enough space message.
Not an issue hence there is a workaround (512K SPIFFS + dropped NetBIOS).

The issue is with uploading index.html.gz on 1M (512K SPIFFS) which gives 470KB space.
Gives me generic upload failed error (not "not enough space" error like with 128K SPIFFS).
Please see the screenshot.
Thanks.
[image]

luc-github, 2019-10-03:
please try to use 194KB for spiffs.

kr15uk, 2019-10-03:
[image]

And serial output
[image]

Steps:
1. sudo esptool.py --chip auto erase_flash
2. Flash with 192K SPIFFS (stock ESP3D, no changes apart uncommented ESP32 line)
3. Boot, connect and [ESP710] for a fresh start
4. Log in hit upload - get an error message
5. [ESP720] -> [ESP710] -> [ESP720]
6. Try another upload - error

Hope this helps.

luc-github, 2019-10-03:
that is weird. I do not have 1M on hand now  but will try with this size on the one I have. web page may have a problem

luc-github, 2019-10-03:
hmmm
I have installed the 2.5.2 core - commented the NETBIOS feature
set the following setting in Arduino IDE
[image]

then compiled and flashed:
`Sketch uses 495882 bytes (59%) of program storage space. Maximum is 827376 bytes.`

then uploaded the index.html.gz

[image]

I cannot reproduce you issue 

I would say it is Puya issue but patch is enabled by default in 2.5.2 - I am dry on this

can you try to upload using https://github.com/esp8266/arduino-esp8266fs-plugin
here the latest: https://github.com/esp8266/arduino-esp8266fs-plugin/releases/download/0.4.0/ESP8266FS-0.4.0.zip

kr15uk, 2019-10-03:
Yep, ESP8266FS did the trick.
[image]

Anyway, I've ordered a couple ESP-01's of AliExpress to see if they'll behave the same way, btw I've also tried to flash couple NodeMCU's with the same settings etc and had no problems there, strange.

Could client software be at the blame?!
I'll power up VM with clean OS to see maybe something to do with my system.

Thanks for all your help and I'll keep you posted on different OS and new ESP's.

Kris

luc-github, 2019-10-04:
Happy you make it - I am not sure what went wrong but it is not the first time I see that an upload using tool do better than a format

closed 2019-10-28 by stale[bot]