luc-github/ESP3D #381: [BUG] The configuration tab does not show any information (empty)

ShadowOfTheDamn, 2019-10-03:
When I open the printer configuration tab there is nothing but the table headers.
the picture is attached.
when I use M503 in the ESP3D console in the web UI I get all of the printer information shown.
M503 Output of esp3d console:

M503
  G21    ; Units in mm (mm)
  M149 C ; Units in Celsius

  M200 D1.75
  M200 D0
  M92 X1600.00 Y1600.00 Z25600.00 E7070.00
  M203 X300.00 Y300.00 Z15.00 E30.00
  M201 X3000.00 Y3000.00 Z100.00 E10000.00
  M204 P3000.00 R3000.00 T3000.00
  M205 B50000.00 S0.00 T0.00 J0.02
  M206 X0.00 Y0.00 Z0.00
  M420 S0 Z0.00
  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
  M301 P13.16 I0.73 D59.21
  M304 P85.56 I16.71 D292.05
  M851 Z-1.20
  M906 X750 Y750 Z750
  M906 I1 Z750
  M906 T0 E400

  M900 K0.00

**Screenshots**
[image]
[image]

**ESP3D Firmware:**
UI: 2.1b20 / FW: 2.1.0.b39
- ESP core version: 2.5.2
- Wifi mode: Station
- WebUI version: 2.1b20
- no change in firmware (config file is attached)
[config.txt](https://github.com/luc-github/ESP3D/files/3687027/config.txt)

**Target Firmware:**
 - Name: Marlin
 - Version 2.0-latest bugfix

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: NodeMCUv2
 - Flash size: 4M: 2M/2M

luc-github, 2019-10-03:
Please update your webui you use the b20 the latest is b51 the b41 has M503 fix. So many update since b20

ShadowOfTheDamn, 2019-10-03:
OK, ty. that did the trick and it is fixed now. I will close the issue.Thanks a lot
can I ask why the latest WebUI is not present in the cloned version of ESP3D from git repo?
best regards

closed 2019-10-03 by ShadowOfTheDamn

luc-github, 2019-10-03:
It is not automatic I think
Need `git submodule update --init`

ShadowOfTheDamn, 2019-10-03:
tnx man, great work.
hope you best.