luc-github/ESP3D #382: [FEATURE REQUEST] Stable host-based printing connected via esp3d-serial (data-port)

p1ngb4ck, 2019-10-04:
Heya =)

As I read multiple times by now, most people seem to use SD-Card-Upload via Wifi only. Unfortunately, for me this is not the desired solution to achieve. What I`d like to get to work is host-based printing from a raspberry-pi running RepetierServer Pro via wifi-serial (I have multiple printers, that shall be controlled by 1 RepetierServer-instance).
Unfortunately, it was impossible for me to get the connection via the data-port stable enough to print.
I have multiple ESP01s and a MKS-TFT-Wifi (ESP12), this is what I achieved so far :
solder bigger EEPRoms to the ESP01s and flash them with esp3d. Flashed the MKS-Wifi with the MKS-Wifi-Firmware. The MKS-Wifi (ESP12) is attached to a MKS-TFT (JGAurora A5), the ESP01s are connected to BTT TFT3.5 v3.0. They connect to the network successfully and I can run simple commands using pronterface, connecting to the data-port.
Unfortunately, I must be doing sth wrong - the serial connection is just unstable. Of course I made sure to have the same baud set on mainboard, tft and esp, I tested 115200 and 250000 baud.

I assume I maybe need to set buffer-sizes and options perfectly correct in RepetierServer, Marlin an ESP. Is there anybody who can shed some more light into this for me please?

Thanks !

luc-github, 2019-10-04:
you can check https://github.com/luc-github/ESP3D/issues/360#issuecomment-522465307

p1ngb4ck, 2019-10-04:
thanks for correcting my request =)

p1ngb4ck, 2019-10-04:
> you can check [#360 (comment)](https://github.com/luc-github/ESP3D/issues/360#issuecomment-522465307)

nice, thanks for pointing this out - will test as soon as I am home, and give some feedback then !

closed 2019-10-28 by stale[bot]