luc-github/ESP3D #383: [FEATURE REQUEST]FTP server

luc-github, 2019-10-11:
Now have mature code for ESP8266 and ESP32 and 3.0 code  is modular it could be good to add ftp server support
also FS could be easily splited at root
/FS/ for flash, /SD/ for SD, /USB/ for USB key and more if any

mat2k, 2019-10-24:
nice. Has there been any progress on that shared SD-card scheme you told me about? Last time I read about it, you told me there were some custom boards due to arrive in a week. i'd definitely like to try something of the sort

luc-github, 2019-10-24:
shared SD card still need special hardware . the current SD implementation in ESP3D is only direct SD connection  so printer has no access to SD

mat2k, 2019-10-24:
yeah, i was asking about the special hardware
i guess you can delete these messages later, I simply think it could be a good idea to make the hardware simple enough for users to DIY

luc-github, 2019-10-24:
I am not electronician the hardware is not done by me but by Roy from panucatt so I cannot really help on this

luc-github, 2019-10-28:
I have added a global FS feature
Flash and SD are virtualy mounted as directories under /
[ESP780] and [ESP790] commands allow to navigate and doing some basic file management
`[ESP780]`
```
Directory on Global FS : /
<DIR> 	FS 	
<DIR> 	SD 	
0 file, 2 dir(s)
```

`[ESP780]/FS`
```
Directory on Global FS : /FS
      	index.html.gz 	6.58 KB 	
1 file, 0 dir
Total 169.38 KB, Used 6.86 KB, Available: 162.51 KB
```
`[ESP780]/SD`
```
Directory on Global FS : /SD
      	myfile.txt 	0 B 	
      	logo2.png 	3.28 KB 	
2 file(s), 0 dir
Total 14.71 GB, Used 64.00 KB, Available: 14.71 GB
```

luc-github, 2019-11-01:
ok feature completed
[image]

FTP transfer is 3X faster compare to web transfer  and using Fatlib is 30% faster

on short transfer speed could change but on long transfer look a like

```
Web Transfert
SDFat lib
1124249 B  =  1.7MB in  21 sec
91327211 B = 87.1MB in  1716 sec = 28min 36s
Speed = 53.2B/s

ESP32 core SD lib
1124249 B =  1.7MB in  37 sec
91327211 B = 87.1MB in  2265 sec = 37 min 45s
Speed = 40,3B/s
 
 
FTP Transfert
SDFat lib
for 91327211 B
Transfer completed in 598710 ms, 152.539979 kbytes/s => 9min 58s
Transfer completed in 578694 ms, 157.816067 kbytes/s => 9min 39s 
 
 
for 1124249 B
Transfer completed in 7579 ms, 148.337380 kbytes/s => 7.6s
Transfer completed in 3821 ms, 294.228998 kbytes/s => 3.8s
Transfer completed in 5586 ms, 201.261905 kbytes/s => 6.6s
```

closed 2019-11-01 by luc-github