luc-github/ESP3D #385: [BUG]<Enter comprehensive title>

goldjunge91, 2019-10-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://github.com/luc-github/ESP3D'
Download Branch 2.1
Unzip 
Open "\ESP3D-2.1\ESP3D-2.1\esp3d.ino"
flash it

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:["2.1.0.b39]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- WebUI version: [e.g. 2.0b4]
- Any change done in FW**  
-- No changes

 **- MCU: [ESP8266] 
 - Name:[D1 mini] 
 - Flash size: [4M]**

**Additional context**
Hi i can not flash the branch 2.1 the compiling error is:
Arduino: 1.8.10 (Windows 10), Board: "Generic ESP8266 Module, 160 MHz, Flash, Disabled, All SSL ciphers (most compatible), ck, 26 MHz, 40MHz, DOUT (compatible), 1M (512K SPIFFS), 2, nonos-sdk 2.2.1 (legacy), v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

Multiple libraries were found for "ESP8266mDNS.h"
In file included from sketch\esp3d.cpp:88:0:

 Used: C:\Users\ps4kk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS
syncwebserver.h:25:30: error: WebSocketsServer.h: No such file or directory

Multiple libraries were found for "ESP8266WebServer.h"
 #include <WebSocketsServer.h>

 Used: C:\Users\ps4kk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer
                              ^

Multiple libraries were found for "EEPROM.h"
compilation terminated.

 Used: C:\Users\ps4kk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\EEPROM
Multiple libraries were found for "DNSServer.h"
 Used: C:\Users\ps4kk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\DNSServer
Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\ps4kk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi
exit status 1
WebSocketsServer.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

luc-github, 2019-10-18:
it is clearly wrote in read me : https://github.com/luc-github/ESP3D#installation

[image]

[image]

goldjunge91, 2019-10-18:
hm sorry but i do not know what to do with it.

goldjunge91, 2019-10-18:
okay i have it.
but that was a bit confusing I have copied them into several folders and compiled again and again until it worked.
you have to put the file in:
C:\Users\"YOURNAME"\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src

closed 2019-10-18 by goldjunge91

luc-github, 2019-10-18:
no this is not right location
should be in my documents\Arduino\libraries

luc-github, 2019-10-18:
[image]

goldjunge91, 2019-10-18:
okay thanks for the information.
by me work the place appdata. and I compiled. it works wonderfully. They did a great job.
I want to cut a USB cable through. and solder the plug to the esp to connect the esp to any printer via usb. is that possible?

luc-github, 2019-10-19:
no it won't esp board  is not an USB host currently, it miss some hardware for this

goldjunge91, 2019-10-19:
hm.. that would be a nice idea for people who do not solder on their mainboard. 
I have seen that the trigorillia board is a pretty tricky job that not everyone can do. then I thought you make a small usb connector and can then connect the esp.

luc-github, 2019-10-19:
yes it is in 3.0 todo list but it is far from easy thing to implement

luc-github, 2019-10-19:
https://github.com/luc-github/ESP3D/issues/239#issuecomment-511144012