luc-github/ESP3D #386: [BUG]ESPCAM-32 cannot get Camera and SD Card

luc-github, 2019-11-01:
I cannot get Camera and SD Card working at once, only one is working when the other is disabled
When both are enabled begining SD MMC give me :
`E (42127) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107`

I though initially it was because Flash led is actually used by SD1 pin and so must not be activated manually, but even disabling CAM_LED_PIN - issue is there.

If I try to init SD before cam - I get SD sometimes working but Webserver and Camserver are no more working and wifi stack is corrupted.

I spent 1 day on this without real success - I am dry on this issue
if anyone has suggestion or even better solution would be great ^_^

ithinkido, 2019-11-02:
https://github.com/jameszah/ESP32-CAM-Video-Recorder

luc-github, 2019-11-02:
yes but it is not helping. it works I know I tested it.  but cannot find why ESP3D is not working with SD MMC begin and camera

luc-github, 2019-11-02:
moving to ESP32 Dev Module board in ide allows to  enable DEBUG
without camera enabled : send [ESP200] after boot
```
;HTTP server started port 80
;SD card detected
```
with camera enabled 
```
;HTTP server started port 80
;Camera server started port 9601
[E][SD_MMC.cpp:84] begin(): Failed to initialize the card (257). Make sure SD card lines have pull-up resistors in place.
;No SD card
```

luc-github, 2019-11-02:
Ok restarting from scratch and seems ok now  for SD and camera but has new problems which should be solved soon.
the positive point diging in it is that allows me to improve debug feature ^_^

luc-github, 2019-11-03:
Ok looks fixed now - camera init process is now fully checked at each step, I have added some hot fixes I found in github and  some from me  to prevent some init error I saw.
Task priority is back to 5 for httpd stream but now it seems not a problem for web server cohabitation. 
I still do not understand the problem I got between SD and Camera streaming - moving back httpd from priority 1 to priority 5 seems part of solution -  I suspect some race condition problem but ...
so 🤞 
I used camera with all settings disabled  but wifi and web server to be sure basic is ok.
Some stress test would be necessary but will do later

luc-github, 2019-11-13:
Looks like the issue I got : https://github.com/espressif/esp-idf/issues/1597#issuecomment-473868610
So issue may still be there

closed 2019-12-07 by stale[bot]

jameszah, 2020-04-16:
I bumped into this discussion while trying to sort out a problem with using a ESP32-CAM with SD, WiFi, and the Camera fighting with each other, and then I discovered I was quoted in the discussion.

My ESP32-CAM-Video-Recorder did not have a problem until I moved to the 1-bit SD access and tried to use the apparently available gpio's 12, 13, and 4.  First I used pin 12 for a PIR, which worked fine, and then started using pin 4 to use the big led (the Blinding Disk-Active Light), which worked fine.  But then I wanted a capacitive Touch switch.  Pin 13 did not work for a Touch (??), so I used pin 12 for Touch, pin 13 for PIR, and pin 4 for the LED, and it all seemed to work ... on my desk.

Then everything went wrong.  The SD card would not mount, and the brownout problems resurfaced on hardware/wires that had worked fine for months.

I think I have sorted it out.  My conclusion (assumption) is that 12, 13, and 4 are used during the SD mount process, even if you are using the 1-bit data access ... pin 13 in particular.  My prototype worked because I was moving in front of the PIR, so there was voltage on pin 13, which is used to enable / power the SD card.  Apparently the SD needs a bunch of current (200 ma ?) to start up, so it had the power, but when I rebooted it when if was not on my desk, and the PIR was not active or not connected, it would fail to mount.  And if the SD did start up normally, then pin 13 would work fine as a PIR, switching on and off without interfering with the SD operations.  

So the moral of this story is to do the power/current hungry activities of wifi startup, sd mount, and camera config with a slight delay so the power supply and capacitors can power everything, and before you touch pins 12, 13, and 4.   Once all the hardware is configured and started, I think you can reconfigure and use those pins -- although I have abandoned those pins for now.