luc-github/ESP3D #387: [BUG] TCP Serial reads corrupted data after a while

TheCrypt0, 2019-11-19:
**Describe the bug**
While printing using the serial TCP interface everything works fine for a while (usually ~1 hour), after that the TCP serial stream starts to read garbage which hangs Simplify3D and ruins the print. 

**NOTE**: After noticing the issue (a few times) I opened the web interface which always worked fine and the commands sent there were executed successfully by the printer (even with the TCP Serial still active and reading garbage). The memory usage was normal: ~40KB free. This might be useful to troubleshoot the issue.

The problem has been observed with both 115200 and 250000 baud.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to ESP3D using [COM2TCP](http://www.astrogeeks.com/AstroGeeks/COM2TCP/index.html) (the issue happens with other TCP serial softwarse too).
2. Start printing via TCP Serial.
3. The issue should happen after ~1 hour.
4. See the following Simplify3D output. You can clearly see where the issue arises.

<details>
  <summary>[SPOILER] Simplify3D Verbose Output</summary>

```
SENT: G1 X116.332 Y86.814 E0.3149
SENT: G1 X116.133 Y86.942 E0.3231
READ: ok
SENT: G1 X115.779 Y87.226 E0.3386
READ: ok
SENT: G1 X115.610 Y87.392 E0.3467
READ: ok
SENT: G1 X115.321 Y87.741 E0.3622
READ: ok
SENT: G1 X115.188 Y87.938 E0.3703
READ: ok
SENT: M105
SENT: G1 X114.975 Y88.338 E0.3859
READ: ok
SENT: G1 X114.886 Y88.555 E0.3939
READ: ok
READ: :200.00 /ok
SENT: G1 X114.813 Y88.777 E0.4019
READ: .00ok
SENT: G1 X114.714 Y89.219 E0.4174
READ: ok
SENT: G1 X114.686 Y89.455 E0.4255
READ: o7 ok
SENT: G1 X114.678 Y89.908 E0.4410
READ: ok0ok
SENT: G1 X114.698 Y90.141 E0.4490
READ: 74
READ: ok:ok
SENT: G1 X114.733 Y90.372 E0.4570
READ: ok
SENT: G1 X114.847 Y90.811 E0.4725
READ: ok
SENT: G1 X114.929 Y91.034 E0.4807
READ: ok
SENT: G1 X115.131 Y91.444 E0.4963
READ: ok
SENT: G1 X115.385 Y91.823 E0.5119
READ: ok
SENT: G1 X115.537 Y92.006 E0.5201
READ: ok
SENT: G1 X115.863 Y92.326 E0.5357
READ: ok
SENT: G1 X116.229 Y92.600 E0.5513
READ: ok
SENT: G1 X116.429 Y92.720 E0.5593
READ: ok
SENT: G1 X116.637 Y92.827 E0.5674
READ: ok
SENT: G1 X117.060 Y92.991 E0.5829
READ: ok
SENT: G1 X117.289 Y93.055 E0.5910
READ: ok
[... 394 lines cut ...]

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: ESP02
 - Flash size: 1M

**Additional context**
Ender 3 Pro running the V1.1.4 Creality Board. 

Marlin configuration updated with increased buffer sizes to handle the increased delay introduced by the TCP communication:
```c++
// The number of linear motions that can be in the plan at any give time.
// THE BLOCK_BUFFER_SIZE NEEDS TO BE A POWER OF 2 (e.g. 8, 16, 32) because shifts and ors are used to do the ring-buffering.
#if ENABLED(SDSUPPORT)
  #define BLOCK_BUFFER_SIZE 64 // SD,LCD,Buttons take more memory, block buffer needs to be smaller
#else
  #define BLOCK_BUFFER_SIZE 64 // maximize block buffer
#endif

// @section serial

// The ASCII buffer for serial input
#define MAX_CMD_SIZE 96
#define BUFSIZE 48
```

luc-github, 2019-11-19:
the tcp bridge is just a bridge so there is not much I could do - I suggest you to set `Include line numbers and checksums` in S3D  (FW Configuration / communication) to avoid data transfert corruption.

TheCrypt0, 2019-11-19:
Thank you @luc-github for taking time to answer.

I'm currently following you suggestion but I hope there's more we can try to do to improve the situation.

The bridge is buffered right? Because it looks like some kind of issue with the ACK by the printer (in ESP3D or possibly in Marlin). As you can see in the S3D output the situation gets worse exponentially ending with a lot of corrupted "ok"s being received. I'm not sure on what is actually going on. 

Printing via USB (115200 or 250000) works fine though.

I suspect that the printer gets the commands successfully but S3D doesn't get the actual "ok" so it waits and hangs the print until a timeout is reached.

luc-github, 2019-11-19:
no, the bridge is actually not buffered https://github.com/luc-github/ESP3D/blob/2.1/esp3d/espcom.cpp#L45

But one question : do you use the webUI when streaming by telnet ?

TheCrypt0, 2019-11-19:
The webUI is closed during the print, I noticed that the latency goes up when it's opened (I guess it's normal since it uses CPU time and networking).

Shouldn't the bridge be buffered to prevent packet loss?

luc-github, 2019-11-19:
it may be buffered at packet level (esp core, lwip) , but not at data level (ESP3D), it is bridged as fast as possible (as soon as received) to avoid delay in tranmission.

Adding the checksum and line numbering will ensure the ok match the line (in theory).
I asked if webui was opened to be sure no communication conflict.
If no communication conflict - there is one bottleneck possible : the buffer you use is may be too big
what size did you defined ?

TheCrypt0, 2019-11-20:
What buffer are you referring to?

luc-github, 2019-11-20:
this one 
[image]

may be try without command buffering at all

TheCrypt0, 2019-11-20:
Disabling the Serial cache on S3D slows down the print to unacceptable levels because it waits for every single "ok" before sending a new command.

I'm currently trying with new buffer sizes both in Marlin and ESP3D, will update if the issue persists.

---

Update: while I was writing the reply the print stopped with the usual garbage received by S3D. The buffers may have helped a lil bit to delay the arise of the issue.
This time I just closed and reopened the virtual COM via S3D and the communication got back to normal.
This could suggests that the issue might  be related to buffers on the computer running the TCP Serial software itself. Further investigation is needed.

I've now switched software to get the TCP Serial data and started a new print. (Currently testing using com0com with com2tcp, an open source alternative to COM2TCP).

luc-github, 2019-11-20:
My suggestion was to see if issue come buffering,  but seems you may found some clue of the  root cause which is better

TheCrypt0, 2019-11-21:
I can confirm that the issue was the Serial TCP emulator (COM2TCP by AstroGeeks).

I've been printing for 24 hours without any issue using [com0com ](http://com0com.sourceforge.net/)with com2tcp (the open source alternative). It's a little bit more complex to setup but I found a pretty handy guide to make it work in less that 10 minutes.

Here's a guide to follow to get the thing set up: http://com0com.sourceforge.net/doc/UsingCom0com.pdf

Thanks @luc-github for the support provided. I'm closing the issue now.

closed 2019-11-21 by TheCrypt0

luc-github, 2019-11-21:
@TheCrypt0 Thank you - that is good thing you shared - may be, would be good to put this in wiki