luc-github/ESP3D #388: [BUG] Strange translation going on with M117=>M147

hapklaar, 2019-11-24:
**Describe the bug**
Whenever I send a M117 command through ESP3D, it seems to translate this to another command, M147.

**To Reproduce**
Steps to reproduce the behavior:
1. Open ESP3d web page
2. Enter 'M117 Testing' into the serial 
3. See output M147 unknown command and S0 unknown command returned from printer (Marlin):

Send: M117 testing123
[...]
Recv: echo:Unknown command: "M147 testing123"
Recv: ok
Recv: echo:Unknown command: "S0"
Recv: ok

**Expected behavior**
'OK' and text on LCD

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- WebUI version: [e.g. 2.0b4]
- Any change done in FW  

**Target Firmware:**
 - Name: Marlin
 - Version 2.0

**Board used (please complete the following information):**
 - MCU: ESP01 
 - Name:
 - Flash size: 4M?

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

luc-github, 2019-11-24:
do you have same problem ?
* when you send command by serial.
* when you send command by telnet (using putty for example).

hapklaar, 2019-11-24:
same issue through esp3d webinterface command input and telnet connection to data port.

When I send M117 to serial of mainboard (SKR 1.3), it works just fine.

luc-github, 2019-11-24:
ok, telnet neither webcommand do text modification

How is your esp board wired to your SKR board ?
What baud rate do you use ?

hapklaar, 2019-11-24:
I updated my OP with how everything is connected. I use 250000 baud, but also tried 115200, no change.

luc-github, 2019-11-24:
you use aux1 so what esp board do you used ? 
Do you mean ESP is connected to TFT and TFT is connected to skr ?
what FW version did you flashed on on ESP3D ?

hapklaar, 2019-11-24:
> you use aux1 so what esp board do you used ?
> Do you mean ESP is connected to TFT and TFT is connected to skr ?
> what FW version did you flashed on on ESP3D ?

That's right. I cloned your repo, compiled and flashed. Versions obtained from web interface:
UI: 2.1b20 / FW: 2.1.0.b39

luc-github, 2019-11-24:
excellent! I just received same screen and esp board yesterday to check some issue I saw on  forums
1 - I cannot reproduce your issue, on my screen when I send `M117 blabla ` I have window on tft which apear with M117 title and blabla text (I have several other issues but may be related to hardware and Marlin/TFT FW) 

2 - as you were able to flash - it means you have serial to USB adapter that you use to flash the esp, please do me a favor and connect your ESP to this serial adapter and open your  serial terminal and send the `M117 blabla` from webUI or telnet to confirm the ESP3D output is correct on serial

hapklaar, 2019-11-24:
Ah yes we are getting somewhere :)

When I try your test, I do see correct output on serial:
[image]

TFT firmware issue?

luc-github, 2019-11-24:
yes definitly :) but I cannot work on this right now sorry - but it is in my plan for next week as several people complained buteven  it is not ESP3D issue I need to dig in - that is why I bought the screen

hapklaar, 2019-11-24:
Great, no rush! I'll keep an eye on this topic.

luc-github, 2019-11-24:
can you share your TFT FW version ?

hapklaar, 2019-11-24:
Mine is latest code built from source. In info screen it identifies itself as:

`Board: BIGTREETECH_TFT35_V3.0
Firmware: TFT35_V3_0.24.2 Nov 23 2019`

I'm going to try that precompiled binary and see if it behaves differently

luc-github, 2019-11-24:
Great, thank you now I know the latest  TFTgit bring more trouble than I found already ^_^

hapklaar, 2019-11-24:
Interesting....

- When I try the precompiled fw BIGTREE_TFT35_V3.0.23.bin (20 sept 2019) , the M117 message is shown on screen.
- Doing the same with precompiled BIGTREE_TFT35_V3.0.24.2.bin (Unified Material Theme - 7 nov 2019) this also still works.
- Recompiled latest src and flashed, and it worked only during startup, showing the Wifi connecting messages on screen ending `Connected' Then followed by:
`Unknown command M147 192.168.111.249`

luc-github, 2019-11-24:
yes definitly in FW TFT - need to check which commit generate such behavior

hapklaar, 2019-11-24:
Tried fork from https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware . This one seems to work ok upon first look. Will need to test more.

luc-github, 2019-11-24:
I think I found the bug reading the code
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L249-L258

the `break`  statement is not in correct position , so next command seems modifying the output and set `4` in 3 char

change the break position from 
```
case 117: //M117
          if (infoMenu.menu[infoMenu.cur] == menuStatus)
          {
            statusScreen_setMsg((u8 *)"M117", (u8 *)&infoCmd.queue[infoCmd.index_r].gcode[5]);
          }
          else
          {
            popupReminder((u8 *)"M117", (u8 *)&infoCmd.queue[infoCmd.index_r].gcode[5]);
            break;
          }
``` 
to
```
case 117: //M117
          if (infoMenu.menu[infoMenu.cur] == menuStatus)
          {
            statusScreen_setMsg((u8 *)"M117", (u8 *)&infoCmd.queue[infoCmd.index_r].gcode[5]);
          }
          else
          {
            popupReminder((u8 *)"M117", (u8 *)&infoCmd.queue[infoCmd.index_r].gcode[5]);
          }
         break;
```

should solve the problem

luc-github, 2019-11-25:
I think issue can be closed 😸

closed 2019-11-25 by luc-github

hapklaar, 2019-11-27:
@luc-github really curious to know how you experience communication through the TFT's ESP port. For me it seems very unstable. For example I only get output from that port after I send something to it, then it stops again. This is very apparent by looking at temperature readings. Setting ESP3d to check position every 1 second helps a bit, but I don't think this is the solution?

luc-github, 2019-11-27:
well I did not test yet so much - it is in my todo list but with the version I have the communication is  only one side (ESP -> TFT ->Printer) for what I can see - I do not get command return even temperatures.
I have tested with a Mega/ramps marlin 2.0 system and tft + esp = reset printer board at start, when no esp is ok, so I suspect power is not enough with 5V avr also I saw repetier was not supported well

So I suspect this TFT is tailored for Bigtreetech board like skr 1.3 but I do not have any and no cash  to buying one for the moment so I plan  to power externaly the TFT to see if power is the root cause and be able to continue.

That said it is not in my top priority list as the issue is not in my FW but in TFT one - I have other topic more urgent to do before going to this sorry.

hapklaar, 2019-11-27:
If you have the motor drivers needed for SKR1.3, like for example tmc2130, I can donate a board to you, no strings attached. It's only around $16 this weekend.

luc-github, 2019-11-27:
Well if you do not have usage of it, this could be useful - yes I should have some tmc2130.
thank you ^_^

hapklaar, 2019-11-27:
If you email me your shipping address, I'll get one to you: hapklaar-at-hapklaar.com

luc-github, 2019-11-27:
PM sent ^_^