luc-github/ESP3D #389: Headbed

rieders, 2019-11-29:
Hi
I have recently seen esp3d and flashed an esp with it. So far everything runs with the Marlin firmware. The only problem I still have is that I have no temperature history of Headbed.
I can not control temperature. In the Marlin Firm I set the temperature sensor of the bed and extruder to 1.

I would be glad if you could help me.

greetings
André

luc-github, 2019-11-29:
did you enabled the bed control in webui preferences ?

rieders, 2019-11-29:
Thank you for the fast answer.

I did not see that there is still a setup menu.

Now it's working.

Many thanks for the help. A great project.

 

greetings

luc-github, 2019-11-29:
great

closed 2019-11-29 by luc-github

rieders, 2020-02-10:
Hi there

I am trying to send files via webgui.
Unfortunately the printer answers "wrong format".
How can I send files to the SD card for printing?

Sincerely yours
André