luc-github/ESP3D #390: [Question] Can I see the serial output with arduino ide

jtkDvlp, 2019-12-06:
Hi there,

I need help, please. I try to use esp3d with my arduino mega ramps 1.4. repitier printer, but it is not working. Starting debugging esp3d I tried to use arduino ide serial monitor to see the commands e.g. M115 sent via esp3d web gui. But I dont get anything. Am I wrong with this test or what could be the problem?

thx

luc-github, 2019-12-07:
it depend how you wired everything but if you use Aux1 = Serial to connect ESP and you try to use the usb of Mega  it won't work  serial cannot be shared, one of them will be unresponsive.

to check :
1 - verify both output on serial monitor separatly and not connected to each other
2 - check TX/RX wires are not up side down
3 - to debug connected esp and ramp use telnet instead

jtkDvlp, 2019-12-07:
Thank you for your reply. I already saw the other issues handling this case, but nothing helped. So now I am trying to check if my esp works as expected connecting via putty to com port but I dont get any output. Connecting to the arduino ramp via putty outputs some stuff.

So what could be the problem? Posting a command via esp3d web gui also brings no serial output on putty console :-(

Any ideas?

luc-github, 2019-12-07:
no idea with current info you provided
pp ls open issue and fill template with all information requested

jtkDvlp, 2019-12-07:
I use an esp2866 ESP-01 and connect it via UART GPIO0 ESP-01 Adaptater. I also upload the sketches via thes adapter. Working with another sketch with this setup works and outputs serial infos to console.

closed 2019-12-07 by luc-github