luc-github/ESP3D #391: [BUG] Don´t get serial output

jtkDvlp, 2019-12-07:
**Describe the bug**
Connecting esp2866 with esp3d to serial terminal like putty I dont get any output nor can input. Sending a command via working esp3d web gui wont also get an output on serial console.

Working with another sketch like "esp2866 TestEspApi" works as expected with serial putty console.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup esp2866 with esp3d 2.1
2. Turn on the device, open web gui to test if its working
3. Connect via putty to com port
4. send am command like M115 via web gui

**Expected behavior**
Sending a command via web gui should give the same command input to serial console output, but got nothing.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:**
- ESP core version: not known what you mean maybe
- Version: 2.1.0.b39
- Web UI Version: 2.1.0.b39
- Wifi mode: STA

Actually I am targeting the following setup, but it dont work, so I tried to check the serial communication of esp3d via console, stepping into the above described bug. Serial communication with repitier works as expected.

**Target Firmware:**
 - Name: Repitier
 - Version 1.0.3

**Board used (please complete the following information):**
 - MCU: ESP2866
 - Name: ??
 - Flash size: 1M

See #390 for previous communication.

luc-github, 2019-12-07:
what is your esp board?

jtkDvlp, 2019-12-07:
I use an esp2866 ESP-01 and connect it via UART GPIO0 ESP-01 Adaptater. I also upload the sketches via thes adapter. Working with another sketch with this setup works and outputs serial infos to console.

luc-github, 2019-12-07:
So you have access to webUI 
can you screenshot the ESP3D settings 
[image]

When you says is working with putty what info do get when you connect ESP +  UART GPIO0 ESP-01 Adaptater to your PC and send `ESP111]` and `[ESP420]` (be sure putty send `\n`) ?

How do you wire the ESP01S board with Ramps ?

jtkDvlp, 2019-12-07:
ESP Settings:
[image]

I cant send anything it seems connect to does also not work. In this case it is not connected to ramp. But I connect it to aux 1 also tried aux 2 and 3, in each case I dont use arduino usb plug to make sure dont get conflicts.

In the meantime I tried an older version of esp3d "ESP3D-0.9.99". With this version it works, I get output on putty console and connecting to ramp etc. I can load printer settings via web gui. So it seems to be a bug of ESP3d 2.1.

luc-github, 2019-12-07:
reading the output give a clue :
[image]

go to ESP3D settings and enable serial - it should help ^_^

jtkDvlp, 2019-12-07:
Where I have to that?

[image]

[image]

[image]

luc-github, 2019-12-07:
do not change anything in FW  code
this is in WEBUI :
[image]

You have old one I guess

jtkDvlp, 2019-12-07:
Damn, you got it I used UI: 0.9.60LE maybe from old test dont know. Using right version and setting up serial works 👍 

Thank you very much 

ps: what is M117 output message?

luc-github, 2019-12-07:
message that go to printer lcd - so not necessary when printer does not have lcd ^_^

Ok good - I close issue then

closed 2019-12-07 by luc-github