luc-github/ESP3D #392: [Question] Serial USB Bridge

kamahat, 2019-12-08:
I'm using Platformio to build a firmware for an SKR pro 1.1. 
I've installed an ESP01s on the dedicated port and that's fine, nice interface.
The question is : what are the good parametres in plateformio to be able to upload the printer firmware via the USB/Serial bridge ?
I've tested a "telnet IP:8888" and I've got a connection.
But what I need to configure/installe in plateformio to use it ? 

`# upload_port       = /dev/ttyACM0
upload_protocol   = serial
upload_port       = 192.168.x.y:8888`

luc-github, 2019-12-08:
I never tried to flash a board like this and I am a newbie in platformIO I use it for a couple of month only
I do not know if such config will work as usually use IP as port is for OTA
In theory it should work as mentioned here : https://github.com/jeelabs/esp-link/blob/master/UC-FLASHING.md#details-of-built-in-avr-flash-algorithm
But I doubt it work out of the box 

I think  you will get timeout as telnet connection response time is slower than serial

It was in my plan to work on this but it is low priority currently in my huge todo list

Please keep update on your finding it is very interresting

luc-github, 2019-12-28:
@kamahat any update ?

kamahat, 2019-12-28:
I've tried what I can but I've discovered that so far it's not supported on SKR pro. On that board we only have "LPC1768 drop file solution" (that I've not able to get functional yet). I hope that Bigtreetech will make the things evolve regarding firmware update.
Currently I have not time to test the USB bridge on GT2530 but I'm keeping this in my todo list.

luc-github, 2020-01-05:
Thank you
I close issue for now then - please re-open when needed

closed 2020-01-05 by luc-github