luc-github/ESP3D #393: [FEATURE REQUEST]BigTreeTech support

luc-github, 2019-12-11:
I read / get report of and  got several problems with Bigtreetech TFT, I have already offered them some help with ESP3D without answer, so this thread is to sum up open topics.
it is about TFT but most of topic should apply to SKR-PRO and the onboard ESP, but I do not have one.

I originally tried tried Ramps/Mega board without success - the TFT3.5 + ESP3D make the board reset in loop and the TFT I got cannot discuss with the ESP board.

With the SKR 1.3 board I got now (Thank you @hapklaar  ^_^) I can use the TFT3.5 3.0 I have without reset 🎆 
But I had to update the TFT FW (I had the first version Vx.x.23) to get ESP module working to Vx.x.25

Also Initially I got some strange behavior, the TFT do not see the board it always says printer not connected - then I flashed with latest SKR 1.3 FW and got it working - (baud rate are 115400 for ESP, 250000 for TFT/SKR) no long run on going yet

I have the another not connected issue with an older version of TFT3.5 (V1.2) without ESP support I have flashed to Vx.x.25 without getting RX working - with multimeter I get continuity between header pin to MCU but did not checked if pin is correct on MCU so not sure it is hardware/FW issue
That said , when updating TFT3.0 V1.2 I originaly copied the whole content of `Copy to SD Card root directory to update` and the update took the V1.1 FW instead of V1.2 which gave unreadable screen so I put the V1.2 FW and UI was back

I will take time to check the TFT code and try to compile it 

Open Topics:
* About instabilities: 
I will have to use it intensively and so I need to setup a system  : I will check with all spare parts if I can do a printer for this, so I need time

* About Listing TFT SDCard / UDrive:
ESP3D send GCODE to printer (M20) to get SD content and this SD will be the onboard or may be the TFT one depending the Marlin configuration and the wiring, currently need the EXP1/2 connected to allow printer to access it.
This means unless the TFT get specific command to list the card content and print command for them, ESP3D cannot do anything - I did not read code yet so may be such commands exists, if yes I will see how to support them

* About Flash size :
 to use all features of ESP3D ESP01-S with 1M is really tight so considerring use ESP12 with 4M is a good alternative IMHO , I saw some suggestion to change the flash on FB, this is not something I can do - I am very bad in soldering, so SMD components is to difficult 😭 

* About Flashing the SKR board using WiFi :
Currently I do not think it is possible - using the bridge and this would be risky if ther is a network issue during flashing 
   - one solution would be to upload FW to ESP flash then ESP would flash the board - not possible today - need to investigate and need bigger ESP flash then ESP01-0S has 
   - another solution is to upload the FW to the onboard SD and reset the board to start the auto update - I do not know yet if a command is available for that, so need to investigate

* About slow uploads:
Same as for ESP3D - serial connection make upload slow compare to SD direct access, I know marlin has a POC of faster binary upload by serial to SD (Optimized SD binary file transfer option) but so far it is mentionned it is only for binary https://github.com/MarlinFirmware/Marlin/pull/13270
So need either
   -  Wifi SD card 
   - to use  the wifi backpack from Panucatt (https://www.panucatt.com/ProductDetails.asp?ProductCode=WB8266), which would solve the flash size and the GCODE upload but exp1/exp2 connectors are not in proper position and need to be wired which can bring some EMI noise on them, but with small adapter should be possible - I need to investigate this also

In  conclusion many open topics, so please be patient in same time feel free to comment / feedback 😺

luc-github, 2019-12-12:
* about TFT3.5 V1.2 with RX not working
even this screen does not support ESP3D the MCU could because it has several USARTs not connected that could be used
So I have switched USART1 to USART2 by connected wires to pins on MCU, and updating the FW to use USART2 as default port  and it solved the issue on my PC with my USB/serial adapter

then I connected to my printer board which I forget has upside down RX/TX and issue is back
So seems connecting wongly RX/TX kill the RX pin ... which is strange I would expect the TX damaged not RX may be an STM32 expert could explain this ?
it looks like hardware weakness as on 3.5 V3.0 such issue did not happened

hapklaar, 2019-12-12:
Hmm that's interesting.... I connected my ESP01 the wrong way around the first time on my TFT35 v2. Then reversed it and everything seemed to work ok-ish... Could this have damaed the TFT35 and caused the strange issue I'm seeing? I only get RX on ESP01 after TX.

luc-github, 2019-12-12:
you mean even using serial usb adapter connected to tft esp connector it is not communicating when you send command like `M117 Hello` ?

hapklaar, 2019-12-12:
No this was with ESP01. I tried to connect a serial to USB adapter to ESP in the form of an ESP programmer, but wasn't successful in getting any meaningful communication, only garbage on every baud setting available.

I still have an old Prolific USB-Serial adapter here I could try...

hapklaar, 2019-12-12:
This prolific wasn't supported anymore, so used another FTDI adapter. Connected it up RX>TX,TX>RX and GND>GND. Also tried swapping TX and RX but getting nothing.

Maybe I'm missing something here:
[image]

luc-github, 2019-12-12:
connection to ESP01 need at least 7 wires https://github.com/luc-github/ESP3D/wiki/ESP8266-01
be sure your ftdi is in 3.3v mode 
you may need another serial adapter to connect to tft so you can see top to end comminication
and see if the issue is RX or TX

hapklaar, 2019-12-12:
Tested my FTDI on scope, it has 3.3v signal output on TX.

Not getting any output from TFT (it should send continuous temperature output) and 'M117 bla' typed in terminal is not received by TFT.

Really baffled by this not working, as ESP01 with your software is able to communicate at least one way.

Hukuma1, 2019-12-13:
I've got a TFT35 V3 along with the SKR 1.3 Mini v1.2 with the wifi module (ESP01-0S) with ESP3D. If you need any testing on this combination I'm ready to contribute. So far I've just been using simple Gcode commands, never tried printing. Sad to hear the flash size is running out, but curious what else could be done. Is there a way to interface it to Cura and print (or save file OTA)?

luc-github, 2019-12-13:
@Hukuma1 thank you for your proposal
well FW is now over 500K so if you drop the Wifi update, you should keep all features
Some tricks can be done : https://github.com/luc-github/ESP3D/issues/331 but I am afraid it is not enough anymore for full features, you may need to choose between wifi update and notifications.

I personally do not recommend to print wireless due to connection latency,  e.g. using repetier host or pronterface, but some people do and succeed. I do no use cura or any slicer as host but I will give same comment.

There are esp projects doing this on github you can give a try if you want ^_^

luc-github, 2020-01-04:
about About Listing TFT SDCard / UDrive: it could be limted to https://github.com/luc-github/ESP3D-WEBUI/issues/83 and PR in TFT code or may need some addition in ESP3D core TBD

luc-github, 2020-02-21:
due to focus on ESP3D 3.0 WebUI the issue support is tagged for 3.0

luc-github, 2020-02-29:
Quick update: I have just received some Winbond flash and upgraded an old ESP01 with 512KB to 4MB

[image]

I have added the link of video tutorial in https://github.com/luc-github/ESP3D/issues/331

Was not easy but I was able to do it when I am very bad in soldering ... so I will use this module with my tests

luc-github, 2020-04-18:
Ok I am now testing the TFT BTT rev 26 and got some issues |
1 - I got some race issue in ESP3D due for web commands that I fixed I think, and I am testing on all 
2 - TFT do a poilling of several commands : M105 M114 and some others
 a - broadcast answer are not currently catched by webUI - fix  is under investigation, it would avoid to query from UI and just catch the regual answers 
 b  - Temperatures broadcast is not done to ESP port - I have a fix for TFT FW but I would like to be sure it is best solution and I would like to do a global PR for other things
 c - Due to ths permanent poilling the SD Upload always failed because poilling disturb the upload so need a command to turn on/turn off the poilling in BTT FW so may be part of part 2 or I will work  on direct SD upload of TFT but need also a BTT FW to allows to list file and print from webUI so also BTT code changes
3 - The rework ESP not originally from BTT work well ...until they are plugged to TFT - not sure what is the problem but electronic is not my best skill if I have any, I only have one BTT TFT with ESP port and only 1 ESP BTT device so I won't investigate to avoid to fry something - Murphy law is always straight to me, I have alreeady fried one BTT screen serial port by just invert the RX/TX so I prefere to keep device safe.  

Negative points : need to do several update in BTT/ESP3D/WebUI
Positive points : things to do become more clear and now I am more familiar with BTT so entries points should be not too hard to implements

luc-github, 2020-04-19:
Update :
* the ESP01 512KB reworked need more power than new silicon version - and drain too much power to TFT, the result is Serial loose some data and screen flickering when doing wifi operations
I have connected a D1 mini and no issue 
* with D1 and the 4MB flash I can now update the FW but the TFT seems overloaded by updated message : may need to do quiet update

luc-github, 2020-04-19:
Additional change necessary is to disable all commands interpretation during upload in TFT
M107 / temperature GCODE etc...

luc-github, 2020-04-20:
For the testers 
https://github.com/luc-github/BIGTREETECH-TouchScreenFirmware/tree/esp3d-support

I have added  `M20 SD:` and `M20 U:` support to list on board SD and USB stick content, I followed the M20 output format

```
M20 SD:
Begin file list
mycode.gco
/TFT35.CUR/
End file list
ok
```

I will add M23/M24 /M25- M524 command to start/resume/pause/cancel  print 

I am not sure need anything else?

luc-github, 2020-04-20:
I have added M23 and M24 command support for local SD and local USB disk
```
M23 SD:/mycode.gco
echo:Now fresh file: SD:/mycode.gco
File opened: SD:/mycode.gco Size: 41
File selected
ok
M24
ok
```

Test and Feedback would be very valuable ^_^

hapklaar, 2020-04-20:
I've been out of the loop for a while, need to read up a bit to see what I've missed :) 

What configuration do you exactly need testing with?

luc-github, 2020-04-20:
thank you
I have btt tft + esp3d,/esp01 + btt board, 
use webui  terminal command or telnet to launch commands

Alex-TV, 2020-04-20:
Good news. I'm starting a check.

hapklaar, 2020-04-20:
Standard ESP01 (512KB/4Mbit) is still ok?

luc-github, 2020-04-20:
you mean 1Mb  flash?

hapklaar, 2020-04-20:
I thought the standard ESP-01 has 512KB flash? I believe mine don't have 1MB, but I could be mistaken. In any case if you say it needs to be a 1MB model, I'll try and source one :)

luc-github, 2020-04-21:
the btt one is 1MB
the blue boards are 512KB
the black boards are 1MB

luc-github, 2020-04-21:
I have added `M115 TFT` command support to help ESP3D to detect if TFT is on path
if not printer will answer with normal M115 answer
I am not sure what is necessary to put so I put minimum data, I have also moved version to 26.1 to better tracking the changes until the final PR, github link will also be changed to BTT one of course
```
M115 TFT
```
```
FIRMWARE_NAME: BIGTREE_TFT35_V3.0.26.1 SOURCE_CODE_URL:https://github.com/luc-github/BIGTREETECH-TouchScreenFirmware/tree/esp3d-support
Cap:TOOL_NUM:1
Cap:EXTRUDER_NUM:1
Cap:FAN_NUM:1
ok
```

if no TFT 
```
M115 TFT
```
```
FIRMWARE_NAME:Marlin 2.0.5.3 (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
...
```

luc-github, 2020-04-25:
Any feedback on TFT FW ?
Also I have updated the web ui to support properly the broadcasted informations so no need to poll position/temperatures any more it will catch the braodcasted ones in webUI 2.1b60

Pandoriaantje, 2020-04-29:
I've just upgraded to a SKR 1.4 turbo, tft3.5 v3 and btt ESP-01. I loaded it with their esp binary, and while all seems to function propperly, I was expecting to also be able to list and upload to the internal SD, for easy firmware updates . But apparently esp3d-web only lists the TFT SD content.
So this is being worked on it seems?

luc-github, 2020-04-29:
@Pandoriaantje latest status for listing TFT onboard SD and USB disk is here : https://github.com/luc-github/ESP3D-WEBUI/issues/83

Upload is not yet possible

Pandoriaantje, 2020-04-29:
> @Pandoriaantje latest status for listing TFT onboard SD and USB disk is here : [luc-github/ESP3D-WEBUI#83](https://github.com/luc-github/ESP3D-WEBUI/issues/83)
> 
> Upload is not yet possible

Thanks for that. I'll keep an eye on the progress. I'm currently dialing in my marlin config, and its a pain in the b*tt to constantly connect usb or remove sd from the board to update new compiled fw.  Would have been nice to just compile, upload, reboot, and be done with it.

luc-github, 2020-04-29:
you mean upload Marlin FW ? upload is only for GCODE file

Pandoriaantje, 2020-04-29:
> you mean upload Marlin FW ? ESP3D upload is only for GCODE file it use M28/M29 GCODE commands

Yes, i was refurring to over the air firmware updates. Trasfering the binary to the onboard sd, and performing a reset. Would save a ton of time.

luc-github, 2020-05-08:
PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/648 for TFT improvement is now merged: 

I think only miss the upload but does it worth it ? M28/M29 is so slow....

Pandoriaantje, 2020-05-13:
> you mean upload Marlin FW ? ESP3D upload is only for GCODE file it use M28/M29 GCODE commands

I was more thinking along the lines of a "over the air marlin firmware update" just like one would do for the ESP firmware, but straight to the printer (onboars SD in case of my SKR 1.4).  Marlin 2.x does have "M28 B1" for binary transfer.

So i was thinking: transfer binary with "M28 B1", and send M997.

luc-github, 2020-05-13:
@Pandoriaantje please check #327

victornpb, 2021-06-21:
I was going to get a SKR2.0 board with TFT, but I'm wondering if it is compatible with ESP3D.

I currently use a ESP01 with RAMPS+MEGA for wifi control and upload.

I was planning in using a ESP32 CAM for controlling it, but it seems that it is not as simple as hooking TX+RX to TXD/RXD

[image]

Legsmaniac, 2021-06-21:
As long as you identify the correct TX/RX pins and set the correct UART in ESP3D firmware, I don't see why not.