luc-github/ESP3D #394: [BUG]<debug message and >

pipi61, 2019-12-13:
**Describe the bug**
Hi!
On esp32, if select USE_SERIAL_2 then LOG to serial port not work (DEBUG_ESP3D & DEBUG_OUTPUT_SERIAL)
Thank You

luc-github, 2019-12-13:
Yes on 2.1 log is only on Serial

USE_SERIAL_2 is for printer clmmunication

You can change it later in config.h, look for `Serial.` and replace by `Serial2.` for debug

pipi61, 2019-12-13:
in config.cpp, in CONFIG::InitBaudrate(long value) function

#ifdef USE_SERIAL_2
    if (Serial2.baudRate() != baud_rate) {
#ifdef ARDUINO_ARCH_ESP8266
        Serial2.begin (baud_rate);
#else
        Serial2.begin (baud_rate, ESP_SERIAL_PARAM, ESP_RX_PIN, ESP_TX_PIN);
//insert next line
	Serial.begin (baud_rate, ESP_SERIAL_PARAM, ESP_RX_PIN, ESP_TX_PIN); 
#endif
Then debug work on serial, printer work on serial2
Thank You

luc-github, 2019-12-13:
Ho I misunderstood your issue.
Good you figured out

pipi61, 2019-12-13:
...
and recommended, please insert in config.h this comments
after this #define-s ...
//#define USE_SERIAL_1	//Serial 1 use same pins (9/10) as flash D2/D3 and QIO is default mode instead of DIO, 
						//so if user do pay attention when compiling, ESP will go to endless restart
//#define USE_SERIAL_2  //ESP32 Better to use Serial 2 as default output because 
Thank You
Nice days

pipi61, 2019-12-13:
warning,
Serial2.begin (baud_rate, ESP_SERIAL_PARAM, ESP_RX_PIN, ESP_TX_PIN); 
Serial.begin (baud_rate, ESP_SERIAL_PARAM, ESP_RX_PIN, ESP_TX_PIN); 
work only if use default pinout (#define ESP_RX_PIN -1, #define ESP_TX_PIN -1 )
else make undefined state

pipi61, 2019-12-13:
next proposal,
the RESET_CONFIG_PIN define to gpio 0, on development esp boards this gpio pin connected to pushbutton (prog mode select)

luc-github, 2019-12-14:
actually almost each esp32 pins can be used for Serial. it is true I did not put same flexibility for debug as for printer output
yes reset pin can be IO0 but is there
may I know why use Serial 2 for printer ? and Serial for debug ?

pipi61, 2019-12-14:
Hi!
On the esp32 development board use serial0 to usb-serial converter. That's why I need another serial port to printer. Not lucky the same port use to debug messages and to printer. if same port used, the printer receive the debug messages and can cause an error.
If use another port then proposal another define to serial0/1/2 pin for serial*.begin parameters.
ESP_RX_PIN, ESP_TX_PIN, 
ESP_RX2_PIN, ESP_TX2_PIN ....
(sorry for my english)

luc-github, 2019-12-14:
Ok understood ^_^

luc-github, 2020-01-04:
ESP3.0 handle better the debug but I will review this user case

commit bb2b7a8e (2020-11-02): Fixes #394

Finally
closed 2020-11-02 by luc-github