luc-github/ESP3D #395: ESP01 don´t connect to Fritzbox

yamimann, 2019-12-14:
What is your question ?
My ESP01 is usable in AP mode under 192.168.0.1 but if i try to connect over client mode to my Fritzbox 7560 it doesn´t connect and fallback to AP mode. 
I use an TFT35 V2 with latest firmware and SKR 1.3 with latest Marlin firmware. 
Even when i restart the esp over the GUI i can see that the ESP try to connect and after 30 sec it swithes to the AP mode.
What is my mistake? (Attached are some Infos)
Joerg
[image]
[image]

luc-github, 2019-12-14:
Try this https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino

I suspect your SSID is not well supported - let me know if you can connect to it using this sketch

yamimann, 2019-12-14:
Hello luc,
here is the log from serial monitor:
Connecting to FRITZ!Box 7560 LT
.......
WiFi connected
IP address: 
192.168.178.57
connecting to djxmmx.net:17
sending data to server
receiving from remote server
"A wonderful fact to reflect upon, that every human creature is constituted
 to be that profound secret and mystery to every other."
 Charles Dickens (1812-70)

yamimann, 2019-12-14:
What to do next?
I saw in the past that the ESP connected, i was able to see the IP in the Fritzbox, but if i tried to connect this IP i got a Timeout.
Joerg

luc-github, 2019-12-14:
If you can connected to your AP using the wificlient sketch you should be able to connect using ESP3D

I can imagine wrong connection parameters

Did you typed the SSID in input box  or selected it in search box ?

yamimann, 2019-12-14:
It is selected by search box.

yamimann, 2019-12-14:
I will stop for today.

luc-github, 2019-12-14:
when you will try again put your SSID and password here 
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/wificonf.cpp#L472
if still cannot connect it means one of the others parameters is not suitable for your AP
so comment https://github.com/luc-github/ESP3D/blob/2.1/esp3d/wificonf.cpp#L467
and try again

yamimann, 2019-12-15:
Hello LUC,
i tried to set SSID and PWD in your suggested lines but get error messages. 
I tested everything again with a second ESP01s with the same behavior. So i tested the ESP01 not connected to my TFT35 V2 instead i used the serial adapter for programming connected to a usb power bank. In this case the ESP connected to my Fritzbox and i can play around with the ESP for a short time, after 1 minute the connection get lost and i receive a timeout message if i try to connect to the IP.

luc-github, 2019-12-15:
what error messages did you get ?
So actually your get connection but server kick you out 
can you try this https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino
To check if you have same behavior

yamimann, 2019-12-15:
Hello LUC,
i got this message in Arduino IDE Serial Monitor:
Connected to FRITZ!Box 7560 LT
IP address: 192.168.178.57
MDNS responder started
HTTP server started

But i can not connect to this IP, i will get a Timeout message
Even if i try to ping this IP it is not connected.
??????

luc-github, 2019-12-15:
ok so issue is with your server which does not allow the ESP - this is not related to ESP3D code
Check your router settings you may have some security settings that prevent ESP to stay connected
also be sure you are close to the router when you do test - it also could be a weak signal issue

yamimann, 2019-12-28:
Hello LUC,
if i use the AT Firmware for the ESP01s everything is ok, it connects to the Fritzbox for several hours without any problems. The AP Mode is also ok only the client mode makes problems. I tried several types of the nonos-sdk 2.21 in Arduino IDE. It connects to the Fritzbox, i see the IP adress but if i try to connect to this ip with Firefox browser i get a timeout. I ordered 4mb chips for the ESP to increase the memory. I´m still waiting for delivery.
cu
Jörg

luc-github, 2020-01-05:
@yamimann 
It is interresting it means some setting for IP connection in AT allows ESP to grant access to your Frizbox

as a sample code webserver  which only use SSID / password  does not works - it should be something that someone on https://www.esp8266.com/ may solved - did you check there ?

yamimann, 2020-01-05:
Hello Luc,
thanks for your feedback. I wish you a happy new year. 
What have i done till now: 
I changed the Memorysize to 4 MB by changing the Chip on the ESP01s without success. I get a stable connection to the Fritzbox, i see the IP adress and Datatraffic  in my WLAN Network in the Fritzbox but when i try to connect via firefox and the ip i get a timeout, so no response from the ESP01 like not loading the index.html under this adress.  What to do next? I ordered a D1 mini and programmed it, soldered an adapter for my TFT35 V2.0 and powered it up .... Tata
Everything is fine, i can connect to my printer! Yeah!
Thanks for your help, i think there is a problem with the programming or firmware of the ESP01 the D1 mini works for me.

luc-github, 2020-01-05:
Ho weird - but happy you managed the issue - happy new year to you too ^_^

luc-github, 2020-01-06:
@yamimann did you formated the SPIFFS after changing flash ?

yamimann, 2020-01-06:
Hello Luc,
i programmed the ESP01 after installation of 4 MB chip with the option full erase in 2MB Rom and 2 MB SPIFFS, how can i format the SPIFFS?

luc-github, 2020-01-07:
you can use serial command `[ESP710]FORMAT` it will format the SPIFFS partition, some `not load index.html.gz behavior` can be fixed like this

luc-github, 2020-01-10:
@yamimann  any update?

yamimann, 2020-01-11:
Hello Luc,
i tried everything, i formated it and installed the GUI again without success. I tried to program the ESP01s as D1 mini without success.. my Fritzbox doesn´t like the ESP01 with ESP3d so no problem my D1 mini connects immediately and runs without any problems! So i will stop further investigation.
Joerg

luc-github, 2020-01-11:
Firmware is same for both so should not be it
Now I think I got simillar issue witj one board. I only could connect if I was using fixed IP, for some reason the one gave by DHCP was not working so I set another one

closed 2020-02-04 by stale[bot]