luc-github/ESP3D #397: [Question] Using ESP3D with the BTT SKR Mini E3

Timmiej93, 2019-12-15:
I'm a big fan of the idea of ESP3D. Cheap, simple, DIY. Sadly, I can't get it to work with my SKR Mini E3.

I'm using an ESP-01, which is connected to the TFT header on the SKR board. The 5 volts from the board go through an LDO with some stabilizing capacitors on the output, GND, TX and RX are hooked up. Chip enabled is hooked up to 3.3v. I've also tried hooking up the other 3 pins to 3.3v, didn't change anything.

When I start the printer, the ESP comes alive, and I can access the web interface just fine, but there's no connection to the printer. I've tried swapping the RX and TX cables, but nothing changed. For some reason, the ESP and the SKR board refuse to communicate. Could this be a firmware issue, where the TFT header is inactive, or something like that?

Any help would be greatly appreciated.

luc-github, 2019-12-16:
Hi it is not ESP3D Fw issue for sure ^_^, it is more serial communication problem as you mention

* It could be baud rate not correct on the board I have Printer is 250000 and 115200
* Marlin Firmware need to have the proper serial interface enabled in FW to communicate with ESP3D
* A bug in Marlin FW - I saw some update on some BTT boards to enable communication with ESP3D (https://github.com/MarlinFirmware/Marlin/pull/16211)
* a TFT defective serial  port if wrong connection was done, hope you di not break you port swapping the RX/TX

 you may also have a look here https://github.com/luc-github/ESP3D/issues/393

Timmiej93, 2019-12-16:
Hi, thanks for your reply.

- I've got both baudrates set to 115200, so that should be fine.
- About the serial interfaces, is that what's set with SERIAL_PORT(_2)? If so, SP1 is at -1 for me, SP2 is commented out. Setting either to 0 throws an error (hardcoded), and when I tried setting SP2 to 1, the printer froze up (no boot screen, not responding to anything, just shows the status screen that the printer showed when it was last turned off). Do you have any idea which port I should be using?
- I've looked through that pull request, but other than having the serial wifi interface mentioned in the description, I can't find any reference to it in code. Maybe it's a incorrect description?
- I have switched around TX and RX ports many times, always without any damage, but there's a first time for everything of course. If anything though, I'd expect the ESP to fail, not the board. Let's just hope that's not the case, and that it's software related.

I had already looked through your FR, but I didn't really see anything pop out to me that could be of use. Did I miss something?

luc-github, 2019-12-16:
The issue is that Serial_2 is commented it should be the one for ESP connector 
on my board it must be :
/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1

Timmiej93, 2019-12-16:
Hm, do you have any suggestion for which ports I can try? Like I said, if I set either SP1 or SP2 to 0, I get an error when compiling, and setting SP2 to 1 made my printer freeze up (only solved by flashing a working firmware).

luc-github, 2019-12-16:
did you tried 
```
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
```

Timmiej93, 2019-12-16:
I did, but as you can see [here](https://github.com/MarlinFirmware/Marlin/blob/5103cdcff6a2d40c9d7eb5de2274fe23dbf56797/Marlin/src/HAL/HAL_STM32F1/HAL.h#L81), this file is hardcoded to throw an error when either SERIAL_PORT or SERIAL_PORT_2 is 0. Could it be that I've got a setting wrong, which is making it use the wrong board or something like that?

luc-github, 2019-12-16:
Share your Serial config in Configuration.h like I did

Timmiej93, 2019-12-16:
My current working firmware can be found on my GitHub ([configuration.h](https://github.com/Timmiej93/Marlin/blob/Ender2dev/Marlin/Configuration.h)).

What I've tried:
```#define SERIAL_PORT -1
#define SERIAL_PORT_2 0```
Result:  #error "Serial port 0 does not exist"

```#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1```
Result: Same as above

```#define SERIAL_PORT -1
#define SERIAL_PORT_2 1```
Result: No errors on compile, printer doesn't work

luc-github, 2019-12-16:
did you tried what BTT put in github ?
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/Marlin/Configuration.h
```
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1
```

Timmiej93, 2019-12-16:
Oddly, I hadn't tried that. I was sure that I copied over all changes from those files to my own file.

There's good new and there's bad news though. 
The good news is that after flashing the firmware with SP1: 2, and SP2: -1, I get the IP address of the ESP on the screen of the printer. At this moment, the web GUI is able to control the printer, so it _can_ work.
The bad news is that after a reboot, the printer freezes and doesn't do anything anymore. Also, when I hit the checkmark to automatically check temperatures, a siren-like sound starts coming from my printer.

luc-github, 2019-12-16:
what settings do you have in RX  and TX on advanced configuration ?
does this happen if the esp i not connected ?

Timmiej93, 2019-12-16:
> what settings do you have in RX and TX on advanced configuration ?
I'm not quite sure what you mean, I haven't touched the advanced config file too much. You can check for yourself if that's easier: [configuration_adv.h](https://github.com/Timmiej93/Marlin/blob/Ender2dev/Marlin/Configuration_adv.h)

> does this happen if the esp i not connected ?
Yes, this also happens if the ESP isn't connected.

Another update: I tried some other stuff, like setting the LCD contrast over the GUI, this also results in the siren sound coming from the printer. This also immediately stops the printer from responding to the web GUI, or the LCD encoder knob.

luc-github, 2019-12-16:
Yes sorry I forgot you sent your config
change
`#define TX_BUFFER_SIZE 0`
to 
`#define TX_BUFFER_SIZE 128`
uncomment 
`//#define RX_BUFFER_SIZE 1024`
to 
`#define RX_BUFFER_SIZE 1024`

Nasty things may happen when using Serial without buffer

Timmiej93, 2019-12-16:
Sadly, this didn't change anything. The behavior is still the same: After the initial flash, moving the X,Y,Z works, but using any other function makes the printer freeze up. Rebooting it also makes it freeze. Setting the screen LCD still produces the siren-like sound. 

I did notice that the firmware.bin file didn't change (i.e. the time reported in windows under 'date modified' didn't change), even though I changed the values you mentioned, and rebuilt the firmware, so I completely removed the build folder, rebuilt the firmware, and tried again, but still nothing changed.

luc-github, 2019-12-16:
no idea sorry. did reverting to original FW keep freeze issue ?

Timmiej93, 2019-12-16:
No, thankfully if I flash the firmware that I have on my GitHub page, the printer works fine again. I'll see if I can get some support from BTT, or BTT experts, since the printer just stops working if I set a serial port that's greater than 0, so it can't be related to the ESP3D. Thanks for all the effort you put into helping me, I really appreciate it.

luc-github, 2019-12-16:
Good luck ^_^

closed 2019-12-16 by luc-github

Timmiej93, 2019-12-17:
I honestly can't believe this... I thought "what would happen if instead of using `SERIAL_PORT 2` and `SERIAL_PORT_2 -1` like BTT does, I swap the two around, so it's `SERIAL_PORT -1` and `SERIAL_PORT_2 2`. You know what happens then? It works! 

The printer no longer freezes after checking the temperature, or sending an LCD contrast command, and it also works after a reboot. Yay!

I do still have some questions, but I'll create a new issue for that, to keep everything nice and orderly.

luc-github, 2019-12-17:
which board version do you have ?
1.0 or 1.2 ?

Timmiej93, 2019-12-17:
Well, I ordered the 1.2 version, and have been using the 1.2 base for everything so far, and it says v1.2 on my board, so you can imagine why I didn't try this earlier.

luc-github, 2019-12-17:
Yes confusing

luc-github, 2019-12-17:
At least one issue you solved ^_^