luc-github/ESP3D #399: [BUG] Switched to Marlin 2.x. File upload via WebUI, results in empty uploaded file, but...

MAGIC-SM0KE, 2019-12-16:
_FW version:1.0 # FW target:marlin # FW HW:Serial SD # primary sd:/sd/ # secondary sd:/ext/ # authentication:no_

File uploaded via WebUI, using menu on the right, result in empty uploaded file.
However, from command terminal :  
M110 N0
N1 M28 hello.g*19
N2 M117 hi there*73
N3 M300 P400 S500*33
N4 M29*28

results with successfully uploaded file hello.g (size 31b)

_I've been using EPS3D 1.0 on ESP8266 1Mb, since 2018. Everything worked fine, with Marlin 1.1.9 (8-bit Melzi board). Recently, i switched to SKR 1.3 and SKR 1.3 mini (Marlin 2.x) and this problem appeared._ 

On Marlin Repository [](https://github.com/MarlinFirmware/Marlin/issues/14323)
they mentioned something about a change in M28 functionality, in Marlin 2.0.X

Is this a known problem ? How should i resolve this ?

Thanks

luc-github, 2019-12-17:
ESP3D 1.0 won't work well with Marlin 2.0 need to use ESP3D 2.1 and ESP3D-WEBUI  2.1 which has several fixes for Marlin 2.0

MAGIC-SM0KE, 2019-12-17:
I flashed 2.1 and now, it works great! 
Thanks

closed 2019-12-17 by MAGIC-SM0KE

Praevision, 2020-02-23:
Actually I've got the same problem (UI: 2.1b20 / FW: 2.1.0.b39) on a ESP12F. I've got a Marlin 2.0, which is customized for the SapphirePro (https://github.com/inib/Marlin/tree/2.0.X-SapphirePro-3.5TFT). Manual creation of a file in UI works as described by MAGIC-SM0KE. I tried different sd cards, baud rates, browsers. The result is always a 0 byte file.