luc-github/ESP3D #401: [BUG]ESP3D Won't compile in Arduino

likedabosschris, 2019-12-27:
**Describe the bug**
Here is the Arduino error code I copied:

Arduino: 1.8.10 (Mac OS X), Board: "Generic ESP8266 Module, 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 4MB (FS:2MB OTA:~1019KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Multiple libraries were found for "ESP8266mDNS.h"
In file included from sketch/esp3d.cpp:88:0:
 Used: /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266mDNS
Multiple libraries were found for "ESP8266WebServer.h"
syncwebserver.h:25:30: error: WebSocketsServer.h: No such file or directory
 Used: /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WebServer
 #include <WebSocketsServer.h>
Multiple libraries were found for "EEPROM.h"
                              ^
 Used: /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/EEPROM
compilation terminated.
Multiple libraries were found for "DNSServer.h"
 Used: /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/DNSServer
Multiple libraries were found for "ESP8266WiFi.h"
 Used: /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi
exit status 1
WebSocketsServer.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

**To Reproduce**
Steps to reproduce the behavior:
Hit Export compiled binary
Error pops up.

**Expected behavior**
I expected a .bin file.

**Screenshots**
[image]

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- WebUI version: [e.g. 2.0b4]

**Target Firmware:**
 - Name: Marlin
 - Version 2.0.1

Should note that I have the stock MKS FW 1.0.1 installed presently but I do intend on using Marlin immediately after flashing the wifi module. 

**Board used (please complete the following information):**
 - MCU: ESP8266 
 - Name: ESP-12s 
 - Flash size: 4M

**Additional context**
I am trying to install ESP3D on the MKS Wifi module in the Tevo Nereus. I am using the stock MKS Robin Nano v1.1 Board. I am trying to compile it to binary so I can flash it to the module, as I am not able to connect via USB.

I have tried following the given instructions as close as I can but can't seem to get anywhere due to this error.

luc-github, 2019-12-27:
Did you read the installation part of Readme ?

likedabosschris, 2019-12-27:
The installation in ReadMe is hard to follow. I was reading the installation on GitHub and trying to follow the video.

luc-github, 2019-12-27:
You mean you do not understand the point 2 of installation?

likedabosschris, 2019-12-28:
Not too clear on point 1 or 2, I put the link into Arduino to download the library for the ESP 8266 boards but aside from that I seem to be doing something wrong. I have tried many different things and it doesn't appear that anything I do works.

luc-github, 2019-12-28:
If you don't follow installation Instructions, installation won't be complete, and compilation will fail
Point 2  mention you must manually install some libraries, and your error is a missing library

What you do not understand in point 2?

likedabosschris, 2019-12-28:
Where do I get the library and what all do I need to add? I thought I added all the files.

luc-github, 2019-12-28:
`Add manually libraries present in libraries directory`
From: `libraries`directory of the repository
To : please read arduino documentation  https://www.arduino.cc/en/guide/libraries

likedabosschris, 2019-12-28:
I think I understand now where I misunderstood the libraries, but it is still throwing the exact same error code after adding the libraries to Arduino. I used the library manager, was I supposed to add it via .zip?

luc-github, 2019-12-28:
no you do not understood
the libraries must be copied manually from libraries directory of this repo to your arduino library directory
so no zip and no library manager

QuitButton, 2019-12-28:
I'm getting weird errors trying to compile with 1.8.10 (Windows) too

`Multiple libraries were found for "ESP8266WiFi.h"
syncwebserver.h:25:30: error: WebSocketsServer.h: No such file or directory

 Used: C:\Users\pauls\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi
 #include <WebSocketsServer.h>

Multiple libraries were found for "ESP8266mDNS.h"
                              ^

 Used: C:\Users\pauls\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266mDNS
compilation terminated.

Multiple libraries were found for "ESP8266WebServer.h"
 Used: C:\Users\pauls\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WebServer
Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\pauls\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\EEPROM
Multiple libraries were found for "DNSServer.h"
 Used: C:\Users\pauls\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\DNSServer
exit status 1
WebSocketsServer.h: No such file or directory`

luc-github, 2019-12-28:
@QuitButton same error same root cause

QuitButton, 2019-12-28:
> the libraries must be copied manually from libraries directory of this repo to your arduino library directory

Which I've done, yet I still get the same errors. Do I need to delete the libraries in Arduino15\packages ?

luc-github, 2019-12-29:
@QuitButton 
`C:\Users\pauls\AppData\Local\Arduino15\packages\` is used by Arduino to put boards core not extra libraries 
extra libraries should be put in `C:\Users\pauls\Documents\Arduino\libraries` it should already exists
this path appear in `File > Preferences > Sketchbook location`
[image]
just as mentioned above in arduino documentation https://www.arduino.cc/en/guide/libraries,
and add `\libaries` to this path - forlder should exists
@likedabosschris same for you

then go to my repository folder and copy from the libraries folder the library you are needed
in your case : 
[image]

then copy  the folder needed to arduino libraries folder mentioned earlier

[image]

I cannot explain better than what is described already in manual installation section of https://www.arduino.cc/en/guide/libraries

luc-github, 2020-01-05:
I guess no -  so I close it

closed 2020-01-05 by luc-github