luc-github/ESP3D #403: [FEATURE REQUEST]Add support for MQTT in\out commands

bthome, 2020-01-05:
**Is your feature request related to a problem? Please describe.**
Allow publish\subscribe via MQTT to send and receive status.  For use with HomeAssistant \ NodeRed.

**Describe the solution you'd like**
Expose topics to support SET\GET\STATUS of all the command options and events as printer prints\moves\temp\times\% complete\etc.  Support of HomeAssistant MQTT Discovery (https://www.home-assistant.io/docs/mqtt/discovery/) to expose ESP3D as sensors \ automation.

**Describe alternatives you've considered**
Pooling is an option but would be better to allow the setup of an MQTT broker within the admin UI and define your topic "root".

**Additional context**

Anyway....appreciate all the work.  Just started using it and like what I can do.

luc-github, 2020-01-05:
This is a not a small task to implement this 
already discussed here : https://github.com/luc-github/ESP3D/issues/179#issuecomment-341348614
and current status is here : https://github.com/luc-github/ESP3D/issues/148

I think it can be done one day but do not close futur may be for 3.1.0 or 4.0
I left this open as reference

bthome, 2020-01-31:
Just as a reference, here is a link to an example of a simple implementation that could be used as a basis for integration.  https://github.com/antonyx/ESP32-Serial-Bridge

closed 2020-02-24 by stale[bot]