luc-github/ESP3D #404: need support to a right installation

dadour13, 2020-01-06:
i want informu my feedback.
1st=> i bought on mega2560 +esp8266 inbedded.and one ramps v1.4 
i follow some tutorial, but we don't found the right information to install al thing correctly.
2nd=> which IDE Arduino we need to instel the ESP firmware.
3rd=> after many test i already install marlin 1.1.9 and esp 1.0 and i discovred that we have many update since.
could you please help me?
4th=> after the installation i'm unsatisfied regarding the problem with the screen, it is look like some pin are mixed.
my email id dadour13@gmail.com.
i need to known how and which pin are connected from the module maybe you known?
best regards

luc-github, 2020-01-06:
Sorry I do not know this board check this https://www.instructables.com/id/Arduino-MEGA-2560-With-WiFi-Built-in-ESP8266/
Readme 2.1 is pretty clear for installation and version to be used
What screen do you talk about? What issue?

luc-github, 2020-01-11:
I guess all is Ok then

closed 2020-01-11 by luc-github

dadour13, 2020-01-11:
Yes i found my problems.
It was a bad welding of my screen. I brought a new screen to get the right
information. Readable
Best regards
David Russo

dadour13, 2020-01-11:
Sorry for my late.
So I found my problem. It was from some cold weld.
Means I bought a new screen. 
I will feedback to you when I will swap it. 
Best regards