luc-github/ESP3D #406: [Question] ESP3D only working if RAMPS connected to PC

andresrobinson, 2020-01-10:
I have wired the ESP8622 v3 to the TX/RX ports on AUX1 of my RAMPS 1.4
Baud rate is the same and the ESP8622 is not connected to my PC.
The thing is, when I do connect my RAMPS to the computer, ESP3D works great, but, when I remove the cable the connection is lost.
Anyone knows what could be happening?

I'm running ESP3D latest version, and Marlin 2.0.X.
Marlin 
SERIAL_PORT is set to 0
and SERIAL_PORT_2 is commented.

luc-github, 2020-01-10:
USB and Serial from AUX1 are using same serial, so it should not work together, but only one at once
what you describe is weird because it looks the opposite
What happen if you power board with ESP8266 connected and printer usb disconnected ?
 
 Did you tried  Marlin 1.1.9 to verify it is not a new `feature` of Marlin 2.0.X  and not an hardware issue ? It is working ok with Marlin 1.1.9

andresrobinson, 2020-01-10:
> What happen if you power board with ESP8266 connected and printer usb disconnected ?

The ESP8266 works, it can be accessed but doesn't connect to the printer.

> Did you tried Marlin 1.1.9 to verify it is not a new feature of Marlin 2.0.X and not an hardware issue ?

I haven't tried yet, but I found this https://github.com/MarlinFirmware/Marlin/issues/10219#issuecomment-375999216
Seems like it should work on 2.0.X

luc-github, 2020-01-11:
@andresrobinson  I do not have a setup mega/ramps + Marlin 2.0.x so I cannot check 
let me know if you succeed, but this config works for sure with Marlin 1.1.9 and repetier 1.X

andresrobinson, 2020-01-12:
@luc-github I had a copy of my Marlin 1.1.1 before I upgraded, so I uploaded it to the RAMPS and I didn't worked, I'm starting to think its a ESP8266 problem.

Any hint of the issue could be? I have also tried changing ESP3D serial 0 to 1, no luck with that.

luc-github, 2020-01-13:
@andresrobinson 
so to sum up
* ESP3D communicate with printer and WEBUI works when connected to AUX1 and RAMPS USB connected to PC
* ESP3D do not communicatedwith print but WEBUI works when connected to AUX1 and RAMPS USB not connected to PC
* Same behavior with Marlin 1.1.9 and Marlin 2.0.x

So seems not ESP8266 issue as it works, as expected in point 1, and WebvUI is working on others, looks more wiring issue or hardware issue in RAMPS
 
How did you wired your ESP board ?
what board it is ?
How do you power it ?

andresrobinson, 2020-01-13:
@luc-github 
That is correct indeed.

Here is how I have the board wired up: https://imgur.com/a/RqlFhpf
My board is the exact same one as per the image, an NomeMcu V3.

luc-github, 2020-01-13:
GND is from Aux1?

andresrobinson, 2020-01-13:
No, I'm using an external supply for 5V and GND

luc-github, 2020-01-13:
then it is your issue 
connect your gnd of your power supply and gnd of Aux1

andresrobinson, 2020-01-13:
@luc-github do you like this? https://imgur.com/undefined

I will have a try and let you know

luc-github, 2020-01-13:
esp minimal wiring with printer is RX/TX/GND even with external power of esp

andresrobinson, 2020-01-14:
Thanks for the support, I can confirm it's working on 2.0.X using this wiring:
https://imgur.com/undefined

closed 2020-01-14 by andresrobinson