luc-github/ESP3D #407: [BUG][ESP290] -> Error: Incorrect Command

tmartin007, 2020-01-18:
When employing [ESP290] in a macro in a successless attempt to create pauses between commands, I tried to enter e.g. [ESP290] 3000 into the "Commands" panel in the WebGUI, which yields a "Error: Incorrect Command" response.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the WebGUI
2. Enter [ESP290] 3000 into the "Commands" panel 
3. Click 'Send'
4. See error

**Expected behavior**
1. Go to the WebGUI
2. Enter [ESP290] 3000 into the "Commands" panel 
3. Click 'Send'
4. experience a 3 second delay
5. see "ok"

**[ESP800]
FW version:2.1.0.c26 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81 # hostname:MILL**

Thanks,

luc-github, 2020-01-18:
use latest git of 2.1 branch : [ESP290] was introduced in  27th Aug 2019
your version is from  27th Mar 2019

tmartin007, 2020-01-18:
Thanks for that super-fast response!

I'll give it a shot ...

tmartin007, 2020-01-19:
Hi, 

I have been struggeling with this all day now.
Unfortunately I am experiencing the same problem as a few other poor fellows,
while, strictly following the instructions given on: 

github.com/esp8266/Arduino/blob/master/README.md#installing-with-boards-manager
Installing with Boards Manager

Starting with 1.6.4, Arduino allows installation of third-party platform packages using Boards Manager. We have packages available for Windows, Mac OS, and Linux (32 and 64 bit).

Install the current upstream Arduino IDE at the 1.8.7 level or later. The current version is on the Arduino website <https://www.arduino.cc/en/main/software>.
Start Arduino and open the Preferences window.
Enter https://arduino.esp8266.com/stable/package_esp8266com_index.json into the Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
Open Boards Manager from Tools > Board menu and install esp8266 platform (and don't forget to select your ESP8266 board from Tools > Board menu after installation).

In the meantime, I had even completely deleted the Arduino IDE (and all libs) from my machine and re-installed it from scratch.
So, no other Arduino stuff is installed, except the IDE configured according to the instructions quoted above.

Attached are a couple of screen-shots showing what the folder structure is like and the Arduino IDE Preferrence settings.

Before, I had used the binary provided by BigTreeTech (https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/esp-01s),
which basically worked - except for ESP290 and a few other things that I was hoping might have been fixed in the new version in the meantime.
(I am using a SKR V1.1 Pro with an esp-01s plugged-in.)

BTW, would it be very difficult, to provide a binary for download that I can just flash on a 1MB ESP-01s?

Otherwise, it would be outstanding, if you could point to possible issues that need to be fixed to get the project built.

For your efforts, thank you so much in advance!

Best regards,

-Thomas

Arduino: 1.8.10 (Mac OS X), Board: "Generic ESP8266 Module, 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), no dtr (aka ck), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, NoAssert-NDEBUG, Only Sketch, 115200"

Multiple libraries were found for "ESP8266WiFi.h"
 Used: /Users/tmartin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi
In file included from sketch/esp3d.cpp:88:0:
Multiple libraries were found for "ESP8266mDNS.h"
 Used: /Users/tmartin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266mDNS
Multiple libraries were found for "ESP8266WebServer.h"
 Used: /Users/tmartin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WebServer
syncwebserver.h:25:30: error: WebSocketsServer.h: No such file or directory
Multiple libraries were found for "EEPROM.h"
 #include <WebSocketsServer.h>
                              ^
 Used: /Users/tmartin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/EEPROM
compilation terminated.
Multiple libraries were found for "DNSServer.h"
 Used: /Users/tmartin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/DNSServer
exit status 1
WebSocketsServer.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

luc-github, 2020-01-20:
what issue do you meet you do not mention anything? please open issue with all info so I can see what issue happen

about giving binary there are too many esp board as well as configuration possible so I do not plan to do for the moment.

tmartin007, 2020-01-20:
Hi,

again, thanks for your quick response.

In the meantime, I successfully accomplished the compilation and flashed the ESP module.
Now after entering [ESP290]3000 the command panel shows “Pause” and after 3 seconds “ok”.
This is great!

I would like to offer the following information for users of OSX versions >= 10.14.0 (possibly earlier),
to get the complilation of ESP3D accomplished (and working for a board with 1MB flash size):

1)	Python3 needs to be installed (e.g.) from
	https://www.python.org/ftp/python/3.8.1/python-3.8.1-macosx10.9.pkg

	since it is not pre-installed on the os and obviously a prerequisite

2) 	The “arduinoWebSockets” folder located in  
    /ESP3D-2.1/libraries

    Needs to be copied to: 
    /Users/USER_NAME/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries

3) if using a 1MB ESP board, in Arduino->Tools select:

Flash Size: "1MB (FS:512KB OTA: ~256KB)” 
Debug Level: “NoAssert-NDEBUG” 

I have 3 more issues/questions, though: 

 Issue 1:

I would like to maximize the file-system-space that is available for macros:

When using a 1MB Flash board, you also propose:

remove printf float support - currently not used
got to your ESP core directory to disable float support
type python ./tools/boards.txt.py --nofloat --allgen
be noted this may affect your others ESP sketchs if you disable float support

Question 1:

 Since there is no "ESP core directory” containing a /tools folder with a boards.txt.py (or boards.txt) file,
 after following these instructions: (github.com/esp8266/Arduino/blob/master/README.md# <http://github.com/esp8266/Arduino/blob/master/README.md#>installing-with-boards-manager),
 what needs to be changed, to remove printf float support 

 Issue 2:

In addition, you recommend:

Note for ESP8266 1MB flash : FW is now too big will all features you need to chose strip the FW and select only some features, also WebUI is now also too big for full multilanguage support to fit the 128K SPIFFS so please use pack with limited language (en +another) 
Question 2:

 What needs to be done to yield an index.html.gz that contains english only ?

 Issue 3:

Unfortunately, it seems as if the command (e.g. g-codes) are not forwarded to the SKR V1.1 Pro board.
This worked with the esp3d.bin binary provided by BigTreeTech. 

Question 3:

Do you have any idea what this could be related to and how to fix it ?
(Attached, please find a screenshot of the ESP8266 settings.)

Again, thank you very much in advance!

Best regard,

-Thomas

luc-github, 2020-01-20:
question1: it is present on github at least   
question2: check languages directory 
question3: no screenshot attached so I do not know, you may check BTT suport issue I have created

tmartin007, 2020-01-21:
Hi,

> question1: it is present on github at least
I know, but following the instructions, it is not mentioned that this is (or might be) needed - or how.
I tried copying the directory into the ESP3D project, but was unable to successfully execute the script.

However, it seems that executing this option would only save a few (I assume single-digit ?) KBs ...

> question2: check languages directory
Thanks - that was easy!

> question3: no screenshot attached so I do not know, you may check BTT suport issue I have created

Thanks and sorry: in my case the problem was a BAUD rate mismatch between the Marlin and the ESP3D settings.

Now, that the SKR is receiving commands again, there is still one problem I had not mentioned before:

When I attemtp to execute macros that are a bit longer (say 10 - 20 lines of g-code, or so),
the execution stops at some random line in the middle of the macro (always at the same spot, though),
as if the whole macro text is transmitted to the SKR board in one chunk, but that chunk is too large.
Or maybe it is buffered, but the buffer is not flushed again …

Can you provide any hint what this could be caused by ?

Best regards,

-Thomas

luc-github, 2020-01-22:
question 1 : I have updated the FAQ - this need to know before compiling / flashing in ESP8266 core directory not in ESP3D directory

question 3: Macro just send commands line by line  without waiting ack if put on SPIFFS, for big macro you describe, Marlin may not have enough buffer to get all commands in a row, you may increase buffer, but I suggest to put your macro file on SD it will be properly handled as print file.

closed 2020-02-16 by stale[bot]