luc-github/ESP3D #409: WebSocketsServer.h not found

rainerschulte, 2020-01-30:
Hi Folks...

I have an issue uploading esp3d to my ESP01-S. 
[image]
I tried the "latest" Version 2.1, but found also a 2.5.2 but whithout esp3d.ino inside. 
What needs to be done to get it runnung on Windows 10?
My ArduinoSoftware is Version 1.8.11(Store 1.8.25.0) and I am flashing via USB-Programmer, no own built with Arduino and Jumpercables.

Kind regards

Rainer

luc-github, 2020-01-30:
Did you read the intallation section of read me

rainerschulte, 2020-01-31:
yes.. I read it. I have deleted the old directories ans Zips, downloaded all Stuff from here agein and started from a fresh point. The first Upload via STRG+U worked fine this time, but later when using "ESP Sketch Data Upload" a message is returned: 
---Arduino: 1.8.11 (Windows Store 1.8.28.0) (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DIO, 1MB (FS:512KB OTA:~246KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Users\Rainer\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Users\Rainer\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Users\Rainer\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dio,eesz=1M512,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10811 -build-path C:\Users\Rainer\AppData\Local\Temp\arduino_build_49368 -warnings=none -build-cache C:\Users\Rainer\AppData\Local\Temp\arduino_cache_702981 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mklittlefs.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose D:\Users\Rainer\Downloads\ESP3D-0.6.1\esp8266\esp8266.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Users\Rainer\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Users\Rainer\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Users\Rainer\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dio,eesz=1M512,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10811 -build-path C:\Users\Rainer\AppData\Local\Temp\arduino_build_49368 -warnings=none -build-cache C:\Users\Rainer\AppData\Local\Temp\arduino_cache_702981 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mklittlefs.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose D:\Users\Rainer\Downloads\ESP3D-0.6.1\esp8266\esp8266.ino
Using board 'generic' from platform in folder: D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3
Using core 'esp8266' from platform in folder: D:\Users\Rainer\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3
Detecting libraries used...
[... cut ...] SDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10811 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-ID:\\Users\\Rainer\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\cores\\esp8266" "-ID:\\Users\\Rainer\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\variants\\generic" "D:\\Users\\Rainer\\Documents\\ArduinoData

luc-github, 2020-01-31:
humm 0.6.1 is quiet old and I am even surprised it can compile with 2.6.3 espcore
your issue is with the flash tool of ESP8266 core and I do not use it sorry better ask on esp8266 what's happen
Your Flash is 1M - you better use latest ESP3D version instead of the one your try to use as it has a lot of improvement and bug fix

rainerschulte, 2020-02-01:
Hi..
I shrinked the code with the Python Command line and changed the Debug Mode from None to the last option. Because I have only 1 M. Yesterday I could compile one ESP01-S and had a short Connection to the chip. I wanted to change to DHCP and got an IP-Address, and was able to configure the Web-UI. I htought I could connect it with the Display from my Printer and had the Router SSID and PW set to this box. 
For what reason ever, I wasn't able to get a connect anymore. I think I will try another chip.

luc-github, 2020-02-01:
you mean ESP cannot connect to printer or cannot connect to router ?

MosfetWall, 2020-02-22:
For me this error in arduino IDE was solved by installing this library : 
https://github.com/Links2004/arduinoWebSockets

luc-github, 2020-02-22:
This is explained in intallation section of read me, use the version from libraries directory

rainerschulte, 2020-02-22:
As already mentioned, I have changed the chip and used a new one. Since I hat trouble with SPIFFS Space I have shrinked the Boards-Files again and set my Config in Arduino to 192kb FS/OTA 406kb.
After all was copied succesfull to the chip I configured the Page 192.168.0.1 and later on I got an IP from my Router. I was able to login to the Site and changed the Name of my Printer and put the ESP into the Port fo the TFT35 Display from my SKR 13 Board. Now I need to figure out how to connect and make it working.

rainerschulte, 2020-02-23:
Ok... seems to be a Problem with my MARLIN-Settings or Port from BTT TFT. I will close here for now and continue on BigtreeTech.

closed 2020-02-23 by rainerschulte

jschwalbe, 2020-09-20:
> This is explained in installation section of readme, use the version from libraries directory

I'm having this same issue and I'm reading through `https://github.com/luc-github/ESP3D/wiki/Install-Instructions` but I see no mention of adding any libraries. Am I missing something? Thanks!