luc-github/ESP3D #410: Is it possible to trigger GPIO state using a physical button?

royfocker19, 2020-02-06:
Hello everyone, I added macros to turn on an off the printer using [ESP201]P2 V0 and V1.
This GPIO is connected to the ATX PSON pin. ESP3D is connected using 5VSB and ground from ATX PS. I'm just missing a way to turn on the printer without the WEBUI. Is there a way I'm missing?
Thanks in advance

luc-github, 2020-02-07:
What `[ESP201]P2 V0` do is putting the pin connected to ESP GPIO2 to GND
and `[ESP201]P2 V1` do is putting the pin connected to ESP GPIO2 to HIGH = 3.3v
so you can add a push button connected according your needs

royfocker19, 2020-02-09:
But in that case, if I push the button, it will stay that way, and I won’t be able to do the opposite action using the macro button, right? I was looking for a stateless button, not a switch.

luc-github, 2020-02-09:
I wrote push button not switch https://images.app.goo.gl/yTSEH6QDN4nQAEK86

royfocker19, 2020-02-09:
Hi Luc, but shouldn’t that button maintain the short to keep the ATX alive? What does the ESP201, send just one action or keep it the way you specify with V0 V1?

Press in Dire Situations http://www.nooooooooooooooo.com/

royfocker19, 2020-02-09:
SonI can hardcore something like this on your code, right?
https://www.instructables.com/id/ESP32-With-Capacitive-Touch-Button/

luc-github, 2020-02-09:
this is different but you can try

royfocker19, 2020-02-09:
I know, but my understanding is that you can’t power up an atx psu with a momentary button. You need to maintain the contact between pson and gnd cables to keep in turned on.
Maybe I’m not getting your point on how you would do it with just a button.
Regards
Fede

Press in Dire Situations http://www.nooooooooooooooo.com/

royfocker19, 2020-02-09:
Ok, I won’t be able to test a solution for the next 2 weeks. Do you think that solution I mentioned combined with this https://circuitdigest.com/microcontroller-projects/arduino-interrupt-tutorial-with-examples using the change option could be the right one? If so, I would close the issue.
Regards
Fede

luc-github, 2020-02-10:
no idea , I never tried such thing

royfocker19, 2020-02-26:
I was able to make it work bu the debounce solution is not very good.
Is it possible to call a function to execute a command, for example ESP201?
I was able to serial write but is not the same, I need to call that same action on the code.

luc-github, 2020-02-26:
>Is it possible to call a function to execute a command, for example ESP201?
do you mean : `bool COMMAND::execute_command (int cmd, String cmd_params, tpipe output, level_authenticate_type auth_level, ESPResponseStream  *espresponse)`
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/command.cpp#L138

royfocker19, 2020-02-26:
I was looking that exact line, how could I create a function to call a
specific command, for example [ESP201]P2 V1? I want to add something like
void poweron () {code}

luc-github, 2020-02-26:
check https://github.com/luc-github/ESP3D/blob/2.1/esp3d/syncwebserver.cpp#L940

royfocker19, 2020-02-26:
Ok, I now understand how it works that part but do not know how to pass cmd
value to that function. Thanks anyway.
Even If I was able to do it, my main problem is how to detect debounce and
actual status.

I'm doing this

// variable used for the key press
volatile boolean keyPressed = false;

// variable used for the debounce
unsigned long timeNewKeyPress = 0;
unsigned long timeLastKeyPress = 0;
unsigned int timeDebounce = 1000;

//Setup
void setup()
{
    myesp3d.begin();

    pinMode(pin_LED, OUTPUT);
    digitalWrite(pin_LED,HIGH);
    pinMode(pin_switch, INPUT);
    LEDstatus = digitalRead(pin_LED);

    attachInterrupt( digitalPinToInterrupt(pin_switch), keyIsPressed,
RISING );

}

//main loop
void loop()
{
    LEDstatus = digitalRead(pin_LED);
    if (keyPressed)
     {
          keyPressed = false;
          timeNewKeyPress = millis();

          if ( timeNewKeyPress - timeLastKeyPress >= timeDebounce)
          {
              toggle();
              Serial.print("Toggle"); Serial.println(LEDstatus);
          }
          timeLastKeyPress = timeNewKeyPress;
     }

    myesp3d.process();
}

void keyIsPressed()
{
   keyPressed = true;
   Serial.print("button pressed");
}

void toggle()
{
      LEDstatus = digitalRead(pin_LED);
      if (LEDstatus == LOW) { LEDstatus = HIGH;  } else { LEDstatus = LOW;
}
      digitalWrite(pin_LED, LEDstatus);

}

it works both ways, from web, and with hardware, but wanted to simplify it.

closed 2020-03-21 by stale[bot]