luc-github/ESP3D #412: Connect ESP32 cam to SKR 1.3

abdbeg, 2020-02-10:
Hi

Firstly thank you very much Luc for all your work.

I'm noob and my question is can use ESP32 module with SDcard and Camera: [https://item.taobao.com/item.htm?spm=a230r.1.14.49.f0a564d3jPcJhT&id=604039011170&ns=1&abbucket=15#detail](url) on my SKR 1.3 by using EXP1 and EXP2, here's pinout: 
[image] so my printer can use SD card slot.

Thank you!

luc-github, 2020-02-10:
why do you want to use the ESP32 SD card ? you have one on board already
if you expect to upload a file to ESP32 SD by wifi then use the file on printer by connecting SD pins to printer board - is not that simple - you cannot share SD reader like this

abdbeg, 2020-02-10:
Sorry, my English is bad, do you mean that even if I can connect ESP32 sd card pins to the printer I can use it for upload but the printer will not have access to it? 

I saw that wifi backpack has such a feature, so that's why I'm asking.

luc-github, 2020-02-10:
yes this is what I mean 
wifi backpack has additional hardware to share the sd reader - and custom Firmware also to control the additional hardware

abdbeg, 2020-02-10:
I got it thanks

closed 2020-02-10 by abdbeg

abdbeg, 2020-02-10:
Ah sorry Luc, the last question, about using ESP32 with SD card slot as host, is there any problem with fast printer speeds/reading?

luc-github, 2020-02-10:
>about using ESP32 with SD card slot as host, is there any problem with fast printer speeds/reading?

Sorry I do not understand your question

abdbeg, 2020-02-10:
I mean ESP3D 3.0 has a host capacity which was mentioned here: https://github.com/luc-github/ESP3D/issues/289

So the question is if you can use ESP32 like a host to print something, will there be any problems with high speed prints?

luc-github, 2020-02-10:
>The ESP3D 3.0 will have host capability but I am on begining for the moment

__will__ means in futur - so not yet - so not possible for the moment, sorry so I cannot say 
https://github.com/luc-github/ESP3D/issues/243
https://github.com/luc-github/ESP3D/issues/239#issuecomment-545316712

abdbeg, 2020-02-10:
Yes, I know it will be in future, thanks anyway :)

luc-github, 2020-02-10:
but for fast printing nothing is faster as sd printing - and safer