luc-github/ESP3D #414: bin files?

timbor123, 2020-02-22:
Hello.
If it’s not difficult, please, make the compiled files *bin* for the firmware?
I can not compile the source code for esp07, already a week.

luc-github, 2020-02-22:
there is no plan to deliver binaries as there are too many  compilation target possible, also there are too many configurations settings in ESP3D.

What issues do you meet ?

timbor123, 2020-02-22:
asyncwebserver.cpp:21:22: error: pgmspace.h: No such file or directory
compilation terminated.
exit status 1
pgmspace.h: No such file or directory

luc-github, 2020-02-22:
Please do not use async - it is not well supported currently
I have clarified the readme : 
`Use webserver support (recommended as stable), asyncwebserver support is no more stable on ESP3D`

closed 2020-03-17 by stale[bot]