luc-github/ESP3D #415: How can code responses be received when executed via macro ?

tmartin007, 2020-02-22:
When running macros, it seems "ok" is the "standard" response transmitted to the client.

I have added a proprietary g-code to Marlin (i.e. M880) that activates an air pump to create a vacuum. The implementation also interfaces with an air-pressure sensor via I2C and returns, when a certain threshold vacuum is reached - alternatively, when a timeout is reached.

This code is the last step of a macro that issues a view G0- commands to move a vacuum-grabber close enough to a particular object to be picked up.

The macro is "launched" by an external HTTPClient (running on a small ESP32 board), which would like to receive the measured air-pressure as a return value of the http request that initiates the macro (on a ESP8266 plugged into a SKR 1.1 Pro running Marlin).

Instead, the client only receives a string containing "ok" as a response.

How can I accomplish the desired effect ?

On an unrelated issue (#407), I would like to thank you for your suggestion to increase the command buffers.  - This - eventually - was successful. Since the issue is closed, I cannot report the success in the scope of the original issue (#407). However, the information might be interesting to people who run into similar problems. 

Here is your response dated Jan-22-2020: _Macro currently just send commands line by line without waiting ack if put on SPIFFS, for big macro you describe, Marlin may not have enough buffer to get all commands in a row, you may increase buffer, but I suggest to put your macro file on SD it will be properly handled as print file_

luc-github, 2020-02-22:
>How can I accomplish the desired effect ?

Sorry what is the effect you do expect ?

tmartin007, 2020-02-22:
Thanks for responding so quickly.

M880 returns (“echos”) back the actual air-pressure value as a float,
similar to M114 returning the current positions of the axis.

In the scope of the macro, however, the http client only receives a “ok”
rather than the text echoed by M880 (this would be "expected”).

Best regards

luc-github, 2020-02-22:
the output go to serial and serial output go to websocket so spiffs macro or sd macro
you need to use websocket to get answer

closed 2020-03-22 by luc-github