luc-github/ESP3D #416: [Question]Need to reset after each power cycle

dennisbun, 2020-02-24:
So I've been using ESP 3D for a while and this problem has been around almost since the beginning.

I'm using this [Aliexpress](https://www.aliexpress.com/item/32959497881.html?spm=a2g0o.productlist.0.0.16a1702fdTGljx&algo_pvid=7a464002-1f8a-45c0-bcf2-52f469a392b0&algo_expid=7a464002-1f8a-45c0-bcf2-52f469a392b0-5&btsid=0ab6d70515825574463568579e067a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) for the wifi.
Firmware : Marlin
Board : SKR V1.3
Display : Reprap Discount Smart Controller (2004 LCD)
ESP UI : 2.1b56
ESP FW : 2.1.0.b30

Everything works great except that everytime I restart my machine ( power cycle) the screen just froze and I need to press the reset button, then it boots normally and I can control the machine via wifi. Things to note is that the screen doesn't show the IP address ! 

I've read the guide on how to flash the firmware, but since I have no Idea about the flash size or even if the usb port is working I'm too afraid to try.  One thing I know is that the small button on the side of the usb port is used for the upload mode when held.

Any idea what to do ? Or has anyone used the same ESP ??

luc-github, 2020-02-24:
I do not know this device so I cannot comment on it
now issue can be : 
1 - power issue : esp and lcd draw too much at cold start 
2 - Serial ESP disturb Marlin boot and so your lcd 

can you check if same happen if ESP board rx/tx are not connected , only 3.3V and GND ?

dennisbun, 2020-02-24:
Currently the machine is printing so I might report tomorrow, but I think I've tried this before and if the rx/tx is disconnected everything works fine.

luc-github, 2020-02-24:
No this is only for Marlin installed on ESP32 based board

please uncomment this in config.h of ESP3D:
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/config.h#L97

I think it should solve your issue

dennisbun, 2020-02-24:
Ahhh great tried that and it works as intended now !
I can see the ip address now.

Thanks!

closed 2020-02-24 by dennisbun