luc-github/ESP3D #417: [BUG]Uploading File results in 0 byte files

Praevision, 2020-02-27:
**Describe the bug**
When I upload a gcode file via webUI, only a file with 0 bytes is created on the sd card of the printer. The progress jumps quite immediately to 96% or 99%, no matter how big the file is. After a few seconds a popup appears with the text "Error: 0" (see screenshot). 

Debugging is disabled in ESP3D as well as in Marlin FW. Nothing is attached to the 3d printer.

It's the same issue as described in https://github.com/luc-github/ESP3D/issues/399. But i can't resolve it by updating the firmware, because it's already up-to-date.

**To Reproduce**
Steps to reproduce the behavior:
1. Upload a file via webUI

**Expected behavior**
The file is on the sd card of the 3d printer.

**Things I've already tried**
Different sd cards
Formatting sd cards
Use different baud rates 57600, 115200, 250000
Various gcode files
Different internet browsers to access webUI

**Screenshots**
[image]

**ESP3D Firmware:**
- Version: 2.1.0.b39
- Web UI Version: 2.1b20
- Wifi mode: STA

**Target Firmware:**
 - Name: Marlin
 - Version 2.0 customized for SapphirePro (https://github.com/inib/Marlin/tree/2.0.X-SapphirePro-3.5TFT)

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: ESP12F
 - Flash size: 4M

luc-github, 2020-02-27:
how do you format sd card ?
does listing sd content works  in webui ?

Praevision, 2020-02-27:
I formated it using windows and using the sd card formatter utility.
Listing of sd card works properly. I also can print files which are already present on the sd card via webUI.

luc-github, 2020-02-27:
Can you try to upload this file ?

[test.gco.txt](https://github.com/luc-github/ESP3D/files/4261859/test.gco.txt)

Praevision, 2020-02-27:
> 
> 
> Can you try to upload this file ?
> 
> [test.gco.txt](https://github.com/luc-github/ESP3D/files/4261859/test.gco.txt)

Funnily the upload of this file finished, after having renamed it to test.gco. The content of the file on sd card is the same as of the original file.

luc-github, 2020-02-27:
can you share the file you use and failed?

Praevision, 2020-02-27:
btw. i use Cura 4.4 as slicer:
[Modelle.zip](https://github.com/luc-github/ESP3D/files/4263079/Modelle.zip)

luc-github, 2020-02-27:
nothing pop up on my eyes
can you cut one of the files by half and upload the first part
if failed - cut it by half and try again - until succeed  

need to identify what part failed - a you can see basic upload succeed , so we need more info about failure - if it is esp reset or upload process issue on Marlin (in the past there was a bug with any M290 seen as M29) or ESP serial transfert in ESP3D but was working before, or anything else like instable power supply of esp

Praevision, 2020-02-27:
I tried using your method, but it always failed. So i used your test.gco and just copied the two lines several times. The test.gco with 21kB got uploaded in 10 of 10 attempts. The testB.gco with 28kB got uploaded in 0 of 10 attempts. So it does not depend on the code but on the size. 
[Testcode.zip](https://github.com/luc-github/ESP3D/files/4263626/Testcode.zip)

luc-github, 2020-02-27:
progress bar is depending browser 
what esp board do you use?

Praevision, 2020-02-27:
https://de.aliexpress.com/item/32863572642.html?spm=a2g0s.9042311.0.0.27424c4deKf8Wz --> ESP12F. I'm using it for some other things and never had any problems with it.

luc-github, 2020-02-27:
I need to setup a system to see if I can reproduce

Praevision, 2020-02-27:
[image]

Black: GND
Red: +3V
Orange: RX
White: TX

I actually don't know the board, I'm quite new to the 3d printer topic :D Is it MKS Robin Nano 1.2 as seen in the image? I've got a TwoTrees SapphirePro printer.

luc-github, 2020-02-28:
I have setup a ramps /Mega + GLCD with latest Marlin 2.0.4.4 (TX_BUFFER_SIZE = 128,  RX_BUFFER_SIZE  = 1024 ) , the ESP is an ESP12 E devkit
the wiring is same as one described in Wiki
the upload file is your flexha.gco file, I did it on Windows 10 on Google Chrome
and after 35 min the upload is done as expected. I did twice to double confirm.
[image]
The size difference is due to transfert optimization ignoring the comments
Here video of last try FYR:
[video_of_upload.zip](https://github.com/luc-github/ESP3D/files/4267178/video_of_upload.zip)
 
So 2 guess on your issue:
1 - check your Marlin setting for RX/TX buffer
2 - check your ESP power supply is stable, ESP is very sensitive, if you just wire an ESP8266 mcu as it is, like you linked it,  it may be the root cause of a reset, Serial + WiFi drain lot of power at once, so may be add a capacitor like mentioned in [wiki]( https://github.com/luc-github/ESP3D/wiki/ESP8266-12E-F) would help.

Praevision, 2020-03-02:
Actually you were right. The reason was the power supply. I connected the ESP with a step down regulator directly to the power supply, because i wanted to be able to power off the printer with a relay, and now it's working. Thank you very much for your great support!

closed 2020-03-03 by Praevision