luc-github/ESP3D #419: [Question] NodeMcu v3 board

dmonty2, 2020-02-29:
I have a few questions for my NodeMCU V3  ESP8266 ESP-12E
ESP8266EX
Flash size: 4MB
https://www.aliexpress.com/item/32665100123.html?spm=a2g0s.9042311.0.0.27424c4da70QtM

====
Question 1)  In Arduino IDE the "Flash Size" options are different from the tutorials on the github and youtube.
Tools menu => Flash Size
4MB(FS:2MB OTA:~1019KB)
4MB(FS:3MB OTA:~512KB)
4MB(FS:1MB OTA:~1019KB)
4MB(FS:none OTA:~1019KB)

Which one should I choose?

====
Question 2) mDNS is enabled in the firmware (default)  however It is not advertising to Cura or linux command line avahi-browse -at

luc-github, 2020-02-29:
1- best is 4MB(FS:2MB OTA:~1019KB)
2 - indeed seems mDNS do not broadcast anymore, looking at code seems some update was done a while ago in ESPmDNS library and now need an update() function in loop
I just did a PR to add it https://github.com/luc-github/ESP3D/pull/420 that should solve the problem
thank you for the heads up

dmonty2, 2020-02-29:
Thanks.  You answered and fixed both my questions.

closed 2020-02-29 by dmonty2