luc-github/ESP3D #422: SKR Mini E3 V2.1 - PlatformIO Compiles but will not upload to board

B0M0A0K, 2020-03-07:
I have an SKR-Mini-E3-V2.1 that I have fitted to my Ender 3 Pro. The printer runs just fine with this board but I want to be able to connect to it (using USB provided) in order to flash Marlin to the latest and to enable some features (such as the BLTouch).

I read that in all future updates users will be required to do this via Microsoft Visual Studio Code and the PlatformIO plugin. I have learnt how to do this and the VSC/PlatformIO can build the latest version of Marlin without errors. However, it fails when I try to upload with the following error;

xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

I am using a Windows 10 environment and the MapleDRV is installed and appears to be operating correctly. Pronterface can talk to the board and issue move orders.

I have spent several days trying to diagnose this problem and can see that others are also experiencing the problem. Do you know how to fix this or can you point me to a knowledgebase article that can help please?

luc-github, 2020-03-07:
Is your issue is flashing Marlin on skr?

B0M0A0K, 2020-03-07:
Yes, on SKR-Mini-E3-V2.1

luc-github, 2020-03-07:
so you are in wrong repository
this is not marlin neither bigtreetech repository

luc-github, 2020-03-08:
@B0M0A0K so can issue be closed?

B0M0A0K, 2020-03-08:
Please close.

closed 2020-03-08 by luc-github