luc-github/ESP3D #423: [Question] How to create 8.3 gcode files

Bankinghelp, 2020-03-12:
Since there is no way to upload a normal gcode file with ESP3D and Marlin and it does not automatically convert normals to 8.3 when uploading (which would be a nice feature), I was wondering if there is a way to change the format with cura (changing gcode to gco or g does not work.
Thanks

luc-github, 2020-03-12:
my long file.gcode => longname
myfile.gco = 8.3 format(8 chars no space, extension 3 chars)

>it does not automatically convert normals to 8.3 when uploading (which would be a nice feature),

how to be sure when renaming 8.3 a file is not already existing with this name ? this add extra check / queries and this only for Marlin

8.3 format is a `feature` of Marlin, other Firmwares support long name, I think better to ask Marlin to support long name 😉 (also M33 is not a support of  long name, but an ugly workaround which need a query for each filename and because M20 and other file commands only support 8.3 names)

About cura, I do not use cura so I cannot say what setting to change sorry

luc-github, 2020-03-13:
@Bankinghelp does this answer your question?

Bankinghelp, 2020-03-13:
Yeah, thank u. About the feature I told u, just a tool that when u have selected marlin on ESP3D and upload a gcode, instead of telling u is not possible, the software would just rename it to for example calibrationcube.gcode to calibrei.gco (just an example but would be super useful and easy ^^). I know there is nothing u can do to make marlin support that kind of files but that would be super helpful (and if u knopw how to do it, u can put that in the ESP3D mainpage u see the full name even tho its changed to 8.1)Just Ideas to make this much smoother. Btw thank u for ur time and imma try again just renaming.

luc-github, 2020-03-13:
well from my point of view I just follow Marlin restriction ^_^
also when you upload shortname (8.3) there is no way to display long name as is actually do not exists as it was renamed.
So in that case to do query M33 command to get long name for each name if result is finally a  short name, so no gain but extra work.

Also doing this M33 queries will consume serial communication and result will be slow display, you can imagine a SD card with a lot of files, listing all files names may then take several second and people will complain. 

Now again, renaming could be trouble some if same name already exists(e.g:filenam.gco) so again need extra work to verify that and change name to new name like (filena~1.gco) until name is available .
Finally how end user will know which file is the one he has uploaded? Need to display a message box to tell what name was used? Not really user friendly.

To sum up, there is no easy / good solution. so I prefer the user to be in charge until Marlin support long name like others FW. There is already a PR open for this on Marlin.

I am lazy : I do not fixe what it is not broken. :winkle:

Bankinghelp, 2020-03-13:
kk it's fine, just a suggestion ^^. Thanks!

closed 2020-03-13 by Bankinghelp

Bankinghelp, 2020-03-17:
// This allows hosts to request long names for files and folders with M33
  //#define LONG_FILENAME_HOST_SUPPORT

isnt this what we are looking for? @luc-github

luc-github, 2020-03-17:
did you read my previous comments about M33 command ? M33 must be done for each name

Bankinghelp, 2020-03-17:
ok its fine