luc-github/ESP3D #424: [FEATURE REQUEST] New ESP3D based board

mrv96, 2020-03-15:
Hi, i love your project and since often it's quite annoyng to modify original printers boards in order to get the UART interface (that expecially in economic and spread printers such as Ender 3 don't have a dedicated connector), i'm thinking to develop a new external wifi board.

The new board will be connected through the standard USB port presents in every mainboard and will be equipped with a relay which will let to turn on and off the printer directly via Wi-Fi.
I know that the difficult part is the USB one, but I have already seen two possible solutions.

After my searches i believe that the software to develop is quite easy and will not take too much time, but since I'm a simple electronic engineer i would need a hand expecially because your project is quite big.

If you are intrested in please let me know and contact me. Thank you.

luc-github, 2020-03-15:
it is already in todo list, pls read status here  
https://github.com/luc-github/ESP3D/issues/239#issuecomment-495916269

>After my searches i believe that the software to develop is quite easy and will not take too much time

haha, not so easy per my tests and code evaluation

luc-github, 2020-03-15:
@mrv96 or you have another hardware solution  for the usb host I many not have tested ?

mrv96, 2020-03-15:
The first solution is https://github.com/felis/USB_Host_Shield_2.0, probably you know this library. The main issue is related the missing ch430 driver, but thank to this https://forum.arduino.cc/index.php?topic=647908.0 maybe the work to do is not too much.

I'm not sure about the last sentence, but i asked to you because you are for sure better than me with the software part.

The second solution is this: http://www.hobbytronics.co.uk/usb-host-board-v24. This board can be programmed as a completely transparent bridge between UART and USB slave devices, so in theory there is no necessity to change ESP3D software.
The issue here is that it's quite expensive and even if i want to integrate the hardware in my new board, a lot of people don't have the dedicated PIC programmer to flash the bootloader the first time.

If possible, the first solution is better from the community point of view if not, i will proceed with the second option.

luc-github, 2020-03-15:
That second is definitly interresting, ESP boards have limited capabilities, so adding USB host feature handled by ESP MCU not only add code to ESP3D but also will `eat` some MCU time, I am not even talking about the complexity to support all printer boards and related needed expertise - 

So to sum up : if this can be delegate to a sub board it would be great

About cost : did you do the BOM of this board ? 
price of component is usually lower vs selling  board , 
About pic programmer :  f you populate it on your onwn board you can also flash the bootloader so after there is no issue to update using software described here : http://www.hobbytronics.co.uk/usb-host-board-v24

now back to first solution : the time and expertise spent on sofware/fw are not small
 and as you can see https://github.com/felis/USB_Host_Shield_2.0 and even latest  https://github.com/felis/UHS30 is quite piece of work and still not supporting everything even it is open source for more than 9 years ^_^

 So yes it is possible but need work and time
IMHO  for short term is would evaluate the second solution to see if it is workable and check the limitations (as I am confident not all printer board 's FTDI won't be supported by the host)  and keep the first solution for midlle (if lucky) / long term solution.

mrv96, 2020-03-15:
> About cost : did you do the BOM of this board ?
> price of component is usually lower vs selling board ,
> About pic programmer : f you populate it on your onwn board you can also flash the bootloader [...]

I know that te BOM will be less expensive, but since probably i'll not sell this boards and i'll only release an open-hardware SMD board, so people that are not expert in soldering could have some trouble. If i could use an already existing low cost board i would simplify the soldering process. Patience...

Ok. I'll start to draw the schematic in the next days. Thank you.

luc-github, 2020-03-15:
I see board is still in pre order only so not possible to evaluate 'real' capabilities

mrv96, 2020-03-15:
Are you going to buy a new one to try? If yes i'll wait a feedback from you before to start with the schematic.

luc-github, 2020-03-15:
May be but only when it won't be in pre-order anymore and not before a couple month at least as too busy now with ESP3D 3.0

luc-github, 2020-03-15:
@mrv96 if you start your schematics and want to test your proto board you may try this project : https://github.com/tobozo/ESP32-3DPrinter-Bridge

I was not able to make it work because my printer boards was not recognized. but if you have one that can do,  this is a good start, I have dicussed with developer during my tests and confirmed code  was done to support only one specific FTDI chip

Another reason I could not use it properly also, it is because of the hardware my host boards are clones and usb connector  quality really sucks, making random disconnections  but if you do your own board, and put good connector  - I would be happy to order one from you for my tests 😸

mrv96, 2020-03-15:
Ok. I'll proceed. I'm curious to see what will came out because imo this board can be very useful, also thanks to the relay.

luc-github, 2020-03-15:
yes that is why I did test usb host, I have printers which do not have way to connect ESP boards

SargFowler, 2020-03-15:
A Teensy should be able to interface with the USB port on a printer.
https://www.sparkfun.com/products/13305
Plus, they're quite small and cheap and easy to program.

luc-github, 2020-03-15:
seems there is this lib for teensy 3.6/4.0 https://forum.pjrc.com/threads/58098-Teensy-4-0-USB-Host-Mode

https://www.sparkfun.com/products/13305

mrv96, 2020-03-15:
Mmm... teensy 4.0 cost like the other board and is less convenient from te BOM point of view. IMO the PIC alternative is still the best option. What do you think about?

https://www.sparkfun.com/products/13305

luc-github, 2020-03-15:
well I tested none of the solutions but both have pro and cons IMHO

1 - PIC : 
  pro  : solution may cover all needs and SW is ready
 cons: no source for the software (or I missed it?) so if any issue no easy maintenance

2 - teensy:
   pro : code base is open and available, have a large community
   cons: I do not see any ready solution for our purpose in samples  (or I missed it ?) and so I am not sure it cover all our needs without additional coding, I would say it is close to https://github.com/felis/UHS30 because several PR were done by same developer which is a great contributor in lot of projects

I would say both have good potential, but PIC looks easier at first sight - but I did not do deep investigation...

mrv96, 2020-03-15:
> cons: I do not see any ready solution for our purpose in samples (or I missed it ?) and so I am not sure it cover all our needs without additional coding, I would say it is close to [...]

Looking for CH431 in https://github.com/PaulStoffregen/USBHost_t36/blob/2a3ff1ca872763988df80ad2b66f01ecb73f7a2f/serial.cpp and https://github.com/PaulStoffregen/USBHost_t36/blob/b4956555019850b0a9bcf3050e2aa8afa677e7ca/USBHost_t36.h seems that the 99% of the mainboards could be supported because, differently from UHS2.0 and UHS3.0, it supports the wide spread chinese clone of FTDI CH430G.

Onestly I have good omens about the Teensy solution, but to keep as low as possible the cost of the new board for the moment i'll proceed with the PIC option.

luc-github, 2020-03-16:
another possibility : https://github.com/MatzElectronics/CH559sdccUSBHost

mrv96, 2020-03-16:
> another possibility : https://github.com/MatzElectronics/CH559sdccUSBHost

I can't see any reference to FTDI or something else in the code, so i suppose that cannot be used for our pourpose.

Another possibility is to use a Raspberry Pi Zero. Thanks to its low cost probably is the best solution because it has all the driver that we need, but i never programmed it. Maybe for someone who is confident with Raspberry and Linux in general it could be an easy task.

luc-github, 2020-03-16:
using pi is out of scope, on pi you install octoprint you do not need esp3d then

mrv96, 2020-03-22:
I finished the schematic: https://easyeda.com/mrv96/esp3d-usb

If someone is intrested in the project and wants to try the board i'll try to do the physical PCB in the next days, oherwise i'll make it in a second moment (still undefined).

luc-github, 2020-03-22:
so there is a socket for wemos D1 ?

Did you  tried your schematic with a proto board ?

mrv96, 2020-03-22:
In the PCB there will be a socket for Wemos D1 Mini.

I don't have all the components, so i cannot try anything. Moreover components such as PIC are available only in SMD. To try the schematic is more convenient to produce directly the PCB.

Since the schematic is very simple it should work. The PIC part is copied by the original schematic available on the producer web site.

borland1, 2020-04-10:
Why not just use WEMOS D1 Mini Pro & TSX0102 level shifter for compatibility with both 5V and 3.3V controller processors.  D1 Mini Pro has built-in USB, 16MB flash, and choice of either internal or external WiFi antenna.  TSX0102 is a two channel level shifter, available as breadboard module for $2USD delivered from China, can be mounted on WEMOS D1 Mini shield.  There is also a relay shield if you need that too.

mrv96, 2020-04-11:
A level shifter is not needed, moreover ESP8266 is 5V tolerant, so it can be avoided in the majority of the applications.

The on board USB of D1 mini cannot be used as USB host; it's only a bridge between USB and UART standard.

Regarding relay shields they are great, but since a dedicated board is already required, the relay can be directly mounted on the PCB itself.

luc-github, 2020-04-11:
@mrv96 for the 5V tolerant for long run is would disagree : https://github.com/luc-github/ESP3D/issues/338#issuecomment-526559132
just my 2 cents

mrv96, 2020-04-11:
Ok, i saw only hackdays post. In any case a level shifter is not required at all. The whole boards works at 3.3V. The extenal connectot P4 is placed only for developing and experiments, it shouldn't be used by the end-user.

The main connections with the extern are the 230V and the USB port.

luc-github, 2020-04-25:
@mrv96 did you make any progress ?

mrv96, 2020-04-25:
As i already said the schematic is ready but for the moment, until nobody will try succesfully one of the USB Host proposed method i’m not going to convert the schematic into a PCB. If someone wants to try himself the board that i designed just let me know, i’ll make the PCB in few days.

luc-github, 2020-04-25:
Ok understood - I have to bandwith right now but hope to try it when have time

luc-github, 2020-04-27:
@mrv96 how long would you need to do a board and how much it will cost to send to me in France ?
I cannot do a proto board by myself - I am not skilled ...

bferrarese, 2020-04-27:
I’m willing to try it too. Maybe you send the files and we can each place an order directly with JLCPCB?

luc-github, 2020-04-27:
@bferrarese you mean these files ? https://easyeda.com/mrv96/esp3d-usb

bferrarese, 2020-04-27:
Would needed the Gerber file (used for PCB production) / CPL file(Component Placement List) / PNP file (pick & place file) for SMT assembly.

mrv96, 2020-04-27:
Ok. I'm going to make the PCB. It should be ready at most in 2/3 days. In this way i'll be able to give some prices and the required files for who want to order the board himself.

luc-github, 2020-04-27:
@mrv96 grazie mille 😸

mrv96, 2020-04-28:
> @mrv96 grazie mille 😸

@luc-github prego 🙂

As promised I've just finished the PCB, like the schematic you can find it here: https://easyeda.com/mrv96/esp3d-usb.

[image]

@bferrarese all the file that you need are exportable from everyone directly from EasyEDA and like GitHub you can fork the project and modify it if you want to do someting different.

All the components are available in EasyEDA components store: lcsc.com

Every 3D part excluded the two fuses can be mounted by JLCPCB automatically if wanted.

The most expensive part is the AC-DC converter. Of course it can be avoided if you use an external 5V power supply, but in this way the relay will not let you to turn on and off your printer.

Of course power at the same time Wemon D1 Mini using its USB port and the PCB using its own power supply is not recommended.

I checked again the schematic and i didn't find errors, i hope that the people that will try the board will not have any problem.

P.S. @luc-github tell me if the Wemos relay pin that i chose is ok or if you already use it somewhere in your code please.

luc-github, 2020-04-29:
@mrv96 I think it is ok to use GPIO15 for the moment - code can be modified if necessary, but for Wemos D1 if user want to use the SD card shield with your board, it can be a problem, GPIO15 is the SD card CS for wemos sd card shield, better use GPIO2 if you can in next revision of your board

mrv96, 2020-04-29:
@luc-github It was not a big issue change the GPIO, so i've already put the relay on GPIO2.

Thanks

bferrarese, 2020-04-29:
@mrv96 Awesome! I’ll probably put an order later today. I’ll keep you all posted.

SargFowler, 2020-04-29:
What kind of cost are we looking at for the PCB and all components?

mrv96, 2020-04-29:
I checked the BOM. If you take the components from LCSC, and order non-assembled PCBs from JLCPCB, you’ll pay:

-22.62$ of components for each PCB
-2$ for 5 PCBs (if you panelize the PCB you can get 10 PCBs for 2$)

Shipping cost and Wemos D1 Mini excluded.

WARNING: In order to flash the bootloader on the PIC (only needed one time) it’s needed a PIC programmer (e.g. PicKit 3 or 4).

mrv96, 2020-04-29:
> @mrv96 Awesome! I’ll probably put an order later today. I’ll keep you all posted.

Thank you @bferrarese! If you can post also the total amount spent for who is intrested in.

bferrarese, 2020-04-29:
@mrv96 I'm having an issue. I forked your design, got gerber/bom/pick and place files and I'm trying to place the order but JLCPCB is complaining about two components (X1 and X2) being in the same position. 

```
Repeated coordinate, there are multiple components in the same position.
X1 | 16MHz | HC-49S_L11.4-W4.8  
X2 | 16MHz | OSC-SMD_4P-L3.2-W2.5-BL
```

They do seem to be on top of each other.

Advice?

mrv96, 2020-04-29:
Yes, this is normal. If you are going to mount the PCB yourself you must remove X2 entry from BOM and PICK, otherwise you have to do the same with X1. Only one of the two crystal must be used.

Please note that if you choose to keep X2, since the its pads are under the itself, you will not able to replace/remove it easily by hand.

bferrarese, 2020-04-29:
Alright...removing X1...

I got a quote... 5 boards (that is the minimum) with all available components assembled goes for $69.43 (+shipping).

[image]

It is not really expensive the price per board but I don't really need 5 of those. 
Ideally we need 5 people to split the costs.

mrv96, 2020-04-29:
You can choose to assembly only 2 PCBs at the bottom of the quote page if you want to spend less if i well remember.

[Screenshot from 2020-04-29 18-04-54]<https://user-images.githubusercontent.com/47937307/80655356-0a185f00-8a44-11ea-9875-29c92f181943.png>

bferrarese, 2020-05-26:
No news from me. Like I said, to expensive to buy 5 units alone. @mrv96 is right, you can order only two assembled, but you still buying 5...you get 2 assembled and 3 not assembled, which is kinda worst. I vote to close this one for now.

luc-github, 2020-06-01:
I have got answer from hobbytronics saying they do not plan to release any sources for the FW which can make debug difficult if any issue, they should also have new batch available end of June at best

So I think we can close this topic for now - feel free to open again if any update

closed 2020-06-01 by luc-github