luc-github/ESP3D #425: [Question] Repetier Server Configuration

andresrobinson, 2020-03-18:
I have been able to successfully connect my ESP8266 to my RAMPS running Marlin 2.0
Everything works fine, including heatbed temperature feedback, my Repetier-Host can connect via TCP/IP without problems, but when I use the exact same configuration with Repetier-Server, it takes too long to connect and after it does, It inst possible to send commands, heatbed temperature is 0 and extruder temperature doesn't update.
Is there any special configuration that I might have to use in order for it to work?

luc-github, 2020-03-18:
Sorry I do not use repetier server so cannot help on this - I would suggest to post this on repetier forum to get proper answer, there may be simply a bug in the release of repetier you use

luc-github, 2020-03-22:
@andresrobinson any update ?

luc-github, 2020-04-01:
I guess not

closed 2020-04-01 by luc-github