luc-github/ESP3D #429: [BUG] ESP01S with 1MB Puya flash doesn't work with 2.1.0.b39

Seref, 2020-03-25:
Hey there!

I got an ESP01S with 1MB made by PUYA. I'm using PlattformIO to compile the firmware.
I'm not sure where I exactly fail but I can't ESP3D to run on this ESP, I disabled 
`#define NOTIFICATION_FEATURE`
in config.h and also modified platformio.ini to this:
`[env:esp8266]
platform = espressif8266
board = esp01_1m
framework = arduino
monitor_speed = 115200
; set frequency to 160MHz
board_build.f_cpu = 160000000L
; set frequency to 40MHz
; board_build.f_flash = 40000000L
board_build.flash_mode = dio
upload_resetmethod = ck
build_flags = 
    -Wl,-Teagle.flash.4m2m.ld
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore = 
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD`

I'm not sure why but nothing happens, after erasing the flash and uploading the firmware it doesn
t do anything at all, I don't even get any response over serial. I did found a precompiled firmware for the ESP01S because the PUYA patch isn't applies :(

Is there a correct way for the ESP01S? The current tutorial in the Wiki is a bit old :(

luc-github, 2020-03-25:
for puya patch on 2.5.2 you can add in platformIO.ini : 
`build_flags = -DPUYA_SUPPORT=1`

also I see you select this flash size `4m2m` (4MB flash / 2MB spiffs) but ESP01S has only 1MB flash so it won't work, select (1MB Flash with 192KB SPIFFS)

Seref, 2020-03-25:
Okay that really fixed this! Thanks 
Also I removed 
` -Wl,-Teagle.flash.1m192.ld` from build_flags and added this:
`board_build.ldscript = eagle.flash.1m192.ld` since plattformIO marked the above thing as depracted.

Can I later on add a Wiki for how to setup an ESP01S in PlattformIO? It's a lot more straightforward then I though (though when you set the correct flash sizes)

closed 2020-03-25 by Seref

luc-github, 2020-03-26:
sure, wiki is open