luc-github/ESP3D #430: [Question] Bonjour/hostname.local?

MCUdude, 2020-04-01:
Hi!

First of all, thank you for this great project. The open-source community just keeps on giving!

I have just overhauled my really old (custom) Prusa i3 printer and migrated from an ancient Ramps 1.4 to a brand new 32-bit SKR 1.4 _with_ and ESP-01 socket. I managed to flash the ESP8266 with the ESP3D firmware (had to swap out the flash memory chip for a larger one), and so far everything works great!

One feature request/question though. is it possible to _not_ connect to the printer by using the IP address, but rather a Bounjor hostname like myprinter.local? It's much easier to remember (and type) a hostname plus .local rather than the IP address displayed on the LED on my printer.
I tried to search through the documentation and issues but didn't find anything that could help me figure it out, other that it _might_ be possible.

luc-github, 2020-04-01:
mdns is enabled - use hostname for the name
[image]

Of course your PC need to have bonjour protocol enabled

if you are under windows there is also SSDP enabled 
[image]

MCUdude, 2020-04-01:
Well, my printer's hostname is PRINTER, but I get no response if I try to access (or ping) printer.local. I also downloaded a Bonjour service scanner, and it detected many other devices, but not the ESP8266. Is this something that has to be manually enabled in the firmware source files?

luc-github, 2020-04-01:
what version do you use ?

MCUdude, 2020-04-01:
I downloaded it just a few days ago. It says `UI: 2.1b20 / FW: 2.1.0.b39`

luc-github, 2020-04-01:
So it should be ok, it is enabled by default in FW - I just did a fresh install to double check and it is working on my side
[image]

you can change the hostname to be sure it was not used by another device

in chrome browser, under windows  I type : `myesp3d.local/` and it is reaching the board

MCUdude, 2020-04-01:
That's really weird. I have the same settings as your (mDNS is enabled), but I'm not able to reach the board at prusa.local. I have been using Octoprint in the past, and reaching octopi.local has never been an issue, so I don't think there's anything wrong with my network.

[image]

luc-github, 2020-04-01:
well no, I see you use a recent esp8266 core - I use version 2.5.2

I suggest you to use same version as me to see if it is a core issue

[image]

if still the case please check sample code : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino

ESP3D use exactly same code but it is a good check point to see if it is code or network issue

MCUdude, 2020-04-01:
Thanks! I used PlatformIO to build this. Do you know if there is a way to specify what core version to use? If so, it would be great if this was added to the platformio.ini file for this project.

luc-github, 2020-04-01:
indeed it is not fixed : change this line 
https://github.com/luc-github/ESP3D/blob/2.1/platformio.ini#L38
I tested working - I will double check if it failed with previous value

luc-github, 2020-04-01:
Hmm 2.2.2 is the SDK of NoNOSDK, 2.2.2 is dev version but I set -DNONOSDK221=1
can you share your platformIO.ini ?

MCUdude, 2020-04-01:
Sure!

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir      = esp3d
build_dir    = .pioenvs
lib_dir      = libraries
libdeps_dir  = .piolibdeps
data_dir     = esp3d/data

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
;theboard
board = esp32dev
framework = arduino
monitor_speed = 500000
; set frequency to 240MHz
board_build.f_cpu = 240000000L
; set frequency to 80MHz
board_build.f_flash = 80000000L
board_build.flash_mode = qio
; None
build_flags = -DCORE_DEBUG_LEVEL=0
board_build.partitions = min_spiffs.csv
upload_speed = 921600
lib_ignore = 
    ESPAsyncTCP

[env:esp8266]
platform = espressif8266
board = esp12e
framework = arduino
monitor_speed = 500000
; set frequency to 160MHz
board_build.f_cpu = 160000000L
; set frequency to 40MHz
board_build.f_flash = 40000000L
board_build.flash_mode = dio
upload_resetmethod = nodemcu
build_flags = 
    -Wl,-Teagle.flash.4m2m.ld
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore = 
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD

```

luc-github, 2020-04-01:
then I do not understand how you can get `SDK: 2.2.2-dev(38a443e)`
should be SDK: 2.2.1(cfd48f3) and this is what I have with this platformIO

MCUdude, 2020-04-01:
I did try to recompile by changing the platform line to `platform = espressif8266@2.2.0`.

I uploaded the new firmware using the web interface (browsed for the firmware.bin file). However, [printername].local still doesn't work here for some reason. I'm using the same SDK version as you (SDK:  2.2.1(cfd48f3)).

luc-github, 2020-04-01:
did you tried : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino ?

luc-github, 2020-04-01:
also do you have this line in code ?
https://github.com/luc-github/ESP3D/commit/983d229ee84c742a23854d0cae46db70fe106a69

MCUdude, 2020-04-01:
> also do you have this line in code ?
983d229

Wow, this did the trick. It's working now!

luc-github, 2020-04-01:
ok so you did not have the latest git version actually 
I just did a pre release to be sure the latest is used

closed 2020-04-02 by luc-github