luc-github/ESP3D #431: [Question] Is it possible to run two commands at the same time?

bferrarese, 2020-04-01:
I'm working on a custom UI for a laser cutter and I would like to track machine position while it is playing a file. Is that possible? The main idea is to play a file and have the UI showing machine status like position/limit switches/temperature/etc. Is that possible? Thanks a mil!

luc-github, 2020-04-01:
yes you can it is already the case in current webUI - just do not 'over' query webUI, it can handle one query at once

bferrarese, 2020-04-01:
I guess I'm missing something. Looking at your WEBUI it looks like a regular GET.
If I type "http://myesp3d/command?plain=play%20/sd/mytest_0001.gcode" in a browser window it works as expected and I get the result: 
```
play /sd/mytest_0001.gcode
Playing /sd/mytest_0001.gcode
  File size 21300
```
Now, if while printing I open another tab and enter something like "http://myesp3d/command?plain=M105" I get nothing (timeout). 

What am I missing? Thanks

bferrarese, 2020-04-01:
No heating. Smoothieware.
Its a laser cutter. I have literally no M commands on my gcode. Its a pure gcode.

luc-github, 2020-04-01:
so what command should get with M105 ? it is temperature for 3D printer 
what the output when sending same M105 from serial ?

bferrarese, 2020-04-01:
I have two temperature sensors configured in smoothiware - I for input and O for output... to monitor water cooling temperature. 

this is the regular output of M105
```
ok I:22.9 /0.0 @0 O:82.2 /0.0 @0 
```

On a side note, its not just M105 that timesout. It is any command like M119, M20, etc...

luc-github, 2020-04-01:
Hmm weird - 
you do not wait  between the 2 commands ? 
what happen if you wait during 10 s before sending the second one ?

bferrarese, 2020-04-01:
If I'm not "printing/playing" my UI successfully fires a M114 every second, a M119 and a M220 every 2.5 seconds, and M105 and M130 every 5 seconds. It all works perfectly well if machine is not printing. The moment I issue a "play /sd/..." it all stops. I mean, it plays/prints, but nothing else works until it finish printing.

luc-github, 2020-04-01:
what about :
`http://myesp3d/command_silent?commandText=play%20/sd/mytest_0001.gcode`

bferrarese, 2020-04-01:
Interesting... a slightly different behaviour.

```
http://myesp3d/command_silent?commandText=play%20/sd/mytest_0001.gcode
```

It returns an "ok" almost immediately and prints/play the file. 

Now, while its playing, any other command like

```
http://myesp3d/command?commandText=M119
http://myesp3d/command?commandText=M114
http://myesp3d/command?commandText=M105
http://myesp3d/command?plain=M119
http://myesp3d/command?plain=M114
http://myesp3d/command?plain=M105
```

still comes back empty.

luc-github, 2020-04-01:
the answers are not catched by web but go to websocket

bferrarese, 2020-04-01:
Sorry but you mind clarifying? I do get answers if not playing/printing.

luc-github, 2020-04-01:
I am not sure what is the issue here - I do not have a system smoothieware for laser right now 
 
but my guess is  the handler `command?` wait for an `ok` in answer,  so as soon an ok is sent it  go back
so everything after ok is lost, that is why you must use websocket to catch the full answer

now you wrote if no print command it works well so I am surprised but bug happen.. this is Murphy law...
enabling debug should help to see why the ESP does not answer properly when command , but I think it is due to command print does not send ok 
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/syncwebserver.cpp#L1009-L1136

luc-github, 2020-04-02:
@bferrarese do you have same issue in current webUI?

bferrarese, 2020-04-02:
Hi Luc. Yes, using your stock FW/webUI I have the same issue. I go to the commands panel and type "play xxxx" (I get an ok response and machine starts to play), I type "M114", "M119", "M105", "M20" (any command) and I only get a response after machine finish playing. All responses at once, like they were queued. 

I need to run some errands now but I'll test if same happens using Marlin when I get back.

Thanks for following up.

bferrarese, 2020-04-06:
Hi Luc. I tried a combination of (stock/custom esp fw) x (stock/custom board fw) x (marlin/smoothieware) x (MKS / SKR boards) x (USB / serial / network). I was getting the same result over and over.  It finally hit me... my gcode test file!!! I was using a file with a single move, slow enough to last a few minutes. Everything works as expected with a regular gcode file. :/  
Well, I can't complain...I learned a lot in the process. 
Thanks for your input and willingness to help.

closed 2020-04-06 by bferrarese

luc-github, 2020-04-06:
happy you figured out :)