luc-github/ESP3D #432: Add voltmeter function

jougfr61, 2020-04-06:
I add some resistors between the power supply and the analog pin (ESP8266) and I can read the ADC value by direct command.
Now I would like:
- to convert the digital value (0-255) on analog value : it can be done by a multiplicator value define on config.h (for example ANALOG_CONV 1 by default, or in my case, 0.588) 
- to disp the value on web-UI: add a option to disp this parameter (like DHT sensor) and the Unit (this function can be use for differents analog sensors)

luc-github, 2020-04-06:
need also to do some polling to get the value? what would be the frequency of this polling?

jougfr61, 2020-04-06:
The frequency can be define on the printer menu, like DHT sensor. In my case, each 10 or 30s are enought

luc-github, 2020-08-25:
it is now implemented just need to updated webUI to display it
in configuration.h I have added the convertion function for flexibility

```
//SENSOR_DEVICE: send info based on defined sensor
//DHT11_DEVICE    1
//DHT22_DEVICE    2
//ANALOG_DEVICE   3
//BMP280_DEVICE   4
//BME280_DEVICE   5
#define SENSOR_DEVICE ANALOG_DEVICE

...

#ifdef SENSOR_DEVICE
//pin
#define ESP3D_SENSOR_PIN 22
//Conversion coefficient
#define SENSOR_CONVERTER(v) v*0.588
//Unit to use, if not applicaple for sensor will use default one
//it is used also for the output format
//C for Celsius / F for Fahrenheit / V for volt
#define SENSOR__UNIT "C"
#endif //SENSOR_DEVICE
```

closed 2020-08-25 by luc-github