luc-github/ESP3D #435: No Upload SD Card, Cant Print with ESP3D

adrirock1141, 2020-04-08:
**Describe the bug**
I cant use de SD Card to upload using the WEBIU, I cant Print nothing using the WEBIU, the only thing i can do is home and move axis.
If i try to upload something an error ocurr

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. CLIENT]  
- WebUI version: [e.g. 2.0b4]
- Any change done in FW  

**Target Firmware:**
 - Name: [e.g. Marlin]
 - Version [e.g. 1.1.X]

**Board used (please complete the following information):**
 - MCU: [e.g. ESP8266] 
 - Name:[e.g. NodeMCU V1.0] 
 - Flash size: [e.g. 4M: 2M/2M]

.

luc-github, 2020-04-08:
which error ?

your esp3d version is too old I guess as you do not give it

adrirock1141, 2020-04-08:
> **Describe the bug**
> I cant use de SD Card to upload using the WEBIU, I cant Print nothing using the WEBIU, the only thing i can do is home and move axis.
> If i try to upload something an error ocurr
> 
> **ESP3D Firmware:**
> 
> * ESP core version: 2.5.2
> * Version: 2.1
> * Web UI Version:2.1
> * Wifi mode:CLIENT
> * WebUI version: 2.0
> 
> **Target Firmware:**
> 
> * Name: [e.g. Marlin]
> * Version [e.g. 1.1.X]
> 
> **Board used :**
> 
> * MCU: ESP8266
> * Name:NodeMCU V1.0
> * Flash size: 4M: 2M/2M
> 
> .

> which error ?
> 
> your esp3d version is too old I guess as you do not give it
> 
> left the template without filling information is useless

I just , edit it, sorry im learning how to use this

luc-github, 2020-04-08:
well, take time to collect information, versions, screen shot, etc.

you even do not say what are the error messages look like when uploading or when trying to print, 
sorry but cannot help with  current details

adrirock1141, 2020-04-08:
> well, take time to collect information, versions, screen shot, etc.
> 
> you even do not say what are the error messages look like when uploading or when trying to print,
> sorry but cannot help with current details
Describe the bug
I cant use de SD Card to upload using the WEBIU, I cant Print nothing using the WEBIU, the only thing i can do is home and move axis.
If i try to upload something an error ocur

here i put screen of what happen
[image]
[image]
[image]

Here are the versions of everything

ESP core version: 2.5.2
Version: 2.1
Web UI Version:2.1
Wifi mode:CLIENT
WebUI version: 2.0
Firmware
Name: Marlin
Version 1.1.9
Board
MCU: ESP8266
Name:NodeMCU V1.0
Flash size: 4M: 2M/2M

luc-github, 2020-04-08:
you cannot print such file name in Marlin, only only dos file name are accepted by Marlin.not long name
web ui does not allow such long file name neither but seems you can, which is not normal,
versions you give are incomplete

please copy the output of 
the following commands
[ESP800]
[ESP420]

adrirock1141, 2020-04-08:
[ESP800]
FW version:2.1.0.b39 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81:192.168.0.150 # hostname:Tevo

[ESP420]
{"chip_id":"10464640","cpu":"160","freemem":"36.77 KB","SDK":"2.2.1(cfd48f3)","flash_size":"4.00 MB","update_size":"530.31 KB","spiffs_size":"2.81 MB","baud_rate":"250000","sleep_mode":"Modem","channel":"1","phy_mode":"11g","web_port":"80","data_port":"8888","hostname":"Tevo","active_mode":"STA (68:C6:3A:9F:AD:80)","connected_ssid":"FerNaAd","connected_signal":"86%","ip_mode":"Static","ip":"192.168.0.150","gw":"192.168.0.1","msk":"255.255.255.0","dns":"(IP unset)","disabled_mode":"AP (6A:C6:3A:9F:AD:80)","captive_portal":"Enabled","ssdp":"Enabled","netbios":"Enabled","mdns":"Enabled","web_update":"Enabled","pin recovery":"Disabled","autentication":"Disabled","target_fw":"Marlin","M117_output":"Enabled","Notifications":"Disabled","Serial_output":"Enabled","Websocket_output":"Enabled","TCP_output":"Enabled","fw":"2.1.0.b39 ESP8266/8586"}

adrirock1141, 2020-04-08:
i have read  https://github.com/luc-github/ESP3D/issues/321 and i send you this to complete the information that you may need

M115
FIRMWARE_NAME:Marlin   (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:  EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
M20
Begin file list
/CATANI~1/CATANC~1.GCO 835739
/CATANI~1/CATANC~2.GCO 3413653
/CATANI~1/CATANC~3.GCO 7490159
/CATANI~1/CATANL~1.GCO 3597494
/CATANI~1/CATANP~1.GCO 662555
/CATANI~1/CATANP~2.GCO 3275822
/CATANI~1/CATANP~3.GCO 1377297
/CATANI~1/CATANP~4.GCO 151284
/LIBERA~1/22CALI~1.GCO 43199182
/LIBERA~1/FRAME~1.GCO 89084424
/LIBERA~1/GRIP~1.GCO 31740285
/LIBERA~1/REPUES~1.GCO 4419221
/LIBERA~1/TRIGGE~1.GCO 3598769
/MASCAR~1/1-PEST~1.GCO 745412
/MASCAR~1/2-PEST~1.GCO 1485128
/MASCAR~1/MASCAR~1.GCO 4088894
/MASCAR~1/MASCAR~2.GCO 4088894
/MASCAR~1/TARUGO~1.GCO 2180456
/MASCAR~1/TARUGO~2.GCO 444410
/MASCAR~1/VICERA~1.GCO 2847701
/TEST/MOAI~1.GCO 2413048
/TEST/PATA_T~1.GCO 3216100
/TEST/VIBRAC~1.GCO 6812378
/TEST/CUBO.GCO 362392
HOOK~1.GCO 930620
SEGURO~1.GCO 264488
SOPORT~1.GCO 257554
End file list
M21
echo:SD card ok

luc-github, 2020-04-08:
I think you have issue with webUI as you cannot upload long file name and you show you upload :
`catan ladron.gcode` which is long file name and it is not accepted to be uploaded by webUI, and it is not working with Marlin
you should have this error message:
[image]

adrirock1141, 2020-04-08:
> I think you have issue with webUI as you cannot upload long file name and you show you upload :
> `catan ladron.gcode` which is long file name and it is not accepted to be uploaded by webUI, and it is not working with Marlin
> you should have this error message:
> [image]

i still have this error
[image]

adrirock1141, 2020-04-08:
I just make it work by just 3 letter name and changing the extension of .gcode to .gco
maybe the fact that in my sd cards are folders may cause this error?

luc-github, 2020-04-08:
error 0 is usually due to a reset of esp because of power supply stability
check: https://github.com/luc-github/ESP3D/issues/417

adrirock1141, 2020-04-09:
> error 0 is usually due to a reset of esp because of power supply stability
> check: #417

ok thanks, ill try to put a external source, i have one question more  sorry jejej
there is anyway that ESP3D notifies when the 3d printer finish printing

[image]

i dont know what are these for

adrirock1141, 2020-04-09:
Hello i figured out the notification i read this [https://github.com/luc-github/ESP3D/wiki/Line](url) , the problem i have now it is that when i try to send a message it says, wrong command

[image]

when i send a message it doesnt work
[image]

luc-github, 2020-04-09:
Yes it means the token you have entered is wrong 
also please type `[ESP610]TS=NONE` to clear this ugly line

adrirock1141, 2020-04-09:
> Yes it means the token you have entered is wrong, did you setup it according wiki ?
> also please type `[ESP610]TS=NONE` to clear this ugly line

Yes i follow the wiki
[image]

adrirock1141, 2020-04-09:
i try using this because in Line you only have T1, you dont use T2 and TS
[ESP610]type=LINE T1=xxxxxxxxxxxxxxxxxx T2=NONE TS=NONE
insted of the xxxxxxxxx i put the token of line
but stilll error
[image]

luc-github, 2020-04-09:
what ESP8266 board are you using?

adrirock1141, 2020-04-09:
> what ESP8266 board are you using?

Node MCU V3 Lolin

luc-github, 2020-04-09:
ok because before your wrote `NodeMCU V1.0` so you did not filled proper info
what about webUI version ? you wrote 2.0 and 2.1 but none of them is the full version 
[image]

Also did you do any modification in it ? because  you were able to try to upload long file name which is definitly not possible

adrirock1141, 2020-04-09:
> ok because before your wrote `NodeMCU V1.0` so you did not filled proper info
> what about webUI version ? you wrote 2.0 and 2.1 but none of them is the full version
> [image]
> 
> Also did you do any modification in it ? because you were able to try to upload long file name which is definitly not possible

i dont do any modification, i upload as it was
[image]

luc-github, 2020-04-09:
version 0.9.5!!!! to use with FW 2.1 use UI 2.1....
 so please update it you cannot use such old one use lastest

adrirock1141, 2020-04-09:
how do i update it

luc-github, 2020-04-09:
how did you installed it in first place ?

adrirock1141, 2020-04-09:
> how did you installed it in first place ?

i just did it, i dont installed, i just flash the arduino file and now i read that ESP3D comes with a version but you have tu upload the old one

adrirock1141, 2020-04-09:
Thanks so mucho for your help

adrirock1141, 2020-04-09:
i was talking to early
the same problem 
[image]

[image]

luc-github, 2020-04-09:
the upload is not done by me and neither automatic  - it is done by user as mentioned in https://github.com/luc-github/ESP3D#installation

Anyway now you have latest version and this version works so lets continue...
replace this file (remove the .txt extension - it is to allow to be uploaded) : 
[notifications_service.cpp.txt](https://github.com/luc-github/ESP3D/files/4456459/notifications_service.cpp.txt)
in ESP3D firmware then reflash your board and open arduino serial monitor to see the output
it will show your token so do not share here but verify it is correct
[image]

then in serial monitor type `[ESP600] Hi`

[image]

luc-github, 2020-04-09:
share the output but remove your token - it will says what is wrong

adrirock1141, 2020-04-09:
i figured out, i reflash everything and now it works, i dont know why, but it works

adrirock1141, 2020-04-09:
thanks so much

adrirock1141, 2020-04-09:
its very weird if i dont use a Static Ip it works

adrirock1141, 2020-04-09:
[image]
before i put here a static ip and i put it 192.168.0.150, now that i reflash everything i leave like it was before, and put a static ip in the router
[image]

adrirock1141, 2020-04-09:
these are all the files that de WEBUI has to have or is something else
[image]

luc-github, 2020-04-09:
it may due to your default gateway is not set properly

yes they are, the 2 first are optionals the 3 last are mandatories

closed 2020-04-09 by luc-github