luc-github/ESP3D #436: [FEATURE REQUEST]<NEXTION HMI Screen Integration>

fabian222005, 2020-04-10:
Can you develop with NEXTION HMI SCREEN? Please

luc-github, 2020-04-10:
can you elaborate?

* which version of Nextion?
* for which purpose?

fabian222005, 2020-04-10:
For function like a control panel like in this video [https://youtu.be/Kq1StPYwWfQ](https://youtu.be/Kq1StPYwWfQ) , is for Nextion Enhanced 3.2inch in Nextion editor 1.60.2

luc-github, 2020-04-10:
Yes I know Marlin kimbra support nextion, it is used as display, I have already played with it 

but unless esp act like a host, what is your goal connecting nextion to ESP32 by serial and esp32 to printer by serial ?

I try to understand what would need to be displayed and what would be the usage of touch screen if ESP is not acting as host for streaming GCODE

fabian222005, 2020-04-10:
Yes, for control quickly the printer without another device, like a MakerBot replicator that have an useful screen and a useful app. For a quickly gcode preview like printing time or print dimensions.

luc-github, 2020-04-10:
gcode preview? on Nextion it won't be possible

so nextion is for printer without screen?

fabian222005, 2020-04-11:
Yes, is for a non screen printer like CNC machines

closed 2020-05-05 by stale[bot]