luc-github/ESP3D #437: [FEATURE REQUEST] Add servo or stepper motor integration

fabian222005, 2020-04-10:
Can you add a servo or stepper motor support for change filament color?

luc-github, 2020-04-10:
can you elaborate?

fabian222005, 2020-04-10:
For change filament color like prusa mmu or like [https://youtu.be/0F38WQUEaGs](here) but with a servo an another for another thing. For action a servo using a command like this [ESP500]S:(servo number) D:(Servo degree)

luc-github, 2020-04-10:
Sorry I am not familiar with servo usage so it is not clear for me
 it is not something that is done already  by 3D printer Firmware ? like here https://marlinfw.org/docs/gcode/M280.html

fabian222005, 2020-04-10:
Yes, but I don't like to modify stock hardware or firmware because is a big deal and risk to damage the printer

luc-github, 2020-04-22:
yes but because this feature is already present on printer FW there is no point to add it to ESP3D
 I have no plan to support such feature and I have no bandwith for it, sorry

closed 2020-04-22 by luc-github