luc-github/ESP3D #438: Replace eeprom with w25q128 possible??

tnap1979, 2020-04-12:
So I have a esp-01s with the small memory chip.. 1M I believe..
I have a bunch of RC FPV flight boards around, and one of them that's got a dead stm32f4 chip, has a w25q128 chip on it.. I have a hot air station and chip quick flux, can I just use the 128m chip in place of the w25q32 that is stated... I know it's over kill but it's all I have right now, and with the covid crap... I doubt I'd see my w25q32 chips anytime soon if I ordered them..

luc-github, 2020-04-12:
esp8266 has 16MB board ( D1 mini pro) so w25q128 is supported - just be patient when doing the first time formating of SPIFFS

tnap1979, 2020-04-12:
Sorry for the questions.. lol
How would I go about doing this?
This is my first ever attempt, I know how to use vscode for firmware compilation and arduino ide and such, but never messed chip replacing and reflashing other then my fpv flight controllers

luc-github, 2020-04-13:
I did 4 esp01 with 512KB and put 4MB - I followed this https://www.youtube.com/watch?v=7Q6ABad7U6o&feature=youtu.be

I am not good with a soldering station, I did not used air gun,  I killed one board on the process but the 3 working reworked are ok

[image]

tnap1979, 2020-04-14:
I already have the chip installed, just need to know how to test if it works or not, never really messed with this kind of stuff, coding stuff in arduino, but I just got my raspberry pi 3b+ and camera... so I dunno if I even need to worry about this anymore...
But would still love to know how to test and program it, incase i can still use it for something later

tnap1979, 2020-04-14:
its working fine, all is well

closed 2020-04-14 by tnap1979