luc-github/ESP3D #439: [FEATURE REQUEST] Telegram notification support

fjriosp, 2020-04-13:
It'll be nice if you implement notifications to Telegram.

I've try but I don't know why cannot connect to api.telegram.org:443 using the Notificationclient.

Here you can get all the info about send messages to a Telegram chat via a telegram bot:
https://medium.com/@xabaras/sending-a-message-to-a-telegram-channel-the-easy-way-eb0a0b32968

Create a public channel is optional, really you only need:
1. Create a bot using @BotFather
2. Get the bot token (will be our _token1)
3. Create a chat group with @YourCreatedBot and @RawDataBot
4. Get the chat_id (will be our _token2)
5. Remove @RawDataBot from group

Now you have all the info to make the HTTPS GET request:
https://api.telegram.org/bot[_token1]/sendMessage?chat_id=[_token2]&text=[MY_MESSAGE_TEXT]

Of course, MY_MESSAGE_TEXT must be urlencoded.

Thanks a lot!

luc-github, 2020-04-13:
@fjriosp are you refereing to https://telegram.org/ ?

fjriosp, 2020-04-13:
The only reason is to keep on the group only you and your boy.

RawDataBot is used only to get the chat_id.
You also could create a public channel with a custom name, then you can choice your own chat_id.

fjriosp, 2020-04-13:
Here you have the info about getting the chat_id using RawDataBot.
https://stackoverflow.com/questions/32423837/telegram-bot-how-to-get-a-group-chat-id

luc-github, 2020-04-13:
ho I see I misunderstood the @rawdatabot, I though it was the bot I created
actually I created public channel so I know the name of chatid.

fjriosp, 2020-04-13:
It's ok.
Each user have to configure it's own bot and get the chat_id or create a public channel.
Both options will work.

luc-github, 2020-04-14:
[image]

implemented in 3.0 - but some rewrite will be done, because it use GET request , it need a new `String urlencode(String str)` function for the message text, and I do not know where to put it if need to be reused.
Currently  I have just dropped it in notifications_service.cpp `as it is`, may be try to move to a POST request  would remove the need of it (TBD)
Porting to 2.X may not be done as new feature go to 3.X
Finally I  need to write a proper setup page in Wiki 
I am busy on 3.0  WebUI currently, so I will see when I will do that

fjriosp, 2020-04-14:
Yeah! Thanks a lot!

I'll wait the 3.0 beta to try it.

Thanks!

fjriosp, 2020-04-14:
I'm trying to get it working on 2.1
I'm using an ESP01 board but I'm getting this error on "Notificationclient.connect(_serveraddress.c_str(), _port)":
`M117 Connecting... 
ip:192.168.1.210,mask:255.255.255.0,gw:192.168.1.1
M117 192.168.1.210
State:	sending Client Hello (1)
Alert: handshake failure
Error: SSL error 40
Alert: unexpected message
Error: SSL error 40
Alert: close notify
Error connecting server api.telegram.org:443
pm open,type:2 0
`
Reading on internet it seems a certificate problem. Do you know how to fix it?

luc-github, 2020-04-14:
on esp82866 need to set insecure

fjriosp, 2020-04-14:
Now it's working.
Thanks!

tokaphi, 2020-04-19:
hello, and thank you for your work, 
I just tested and it works very well
it is possible to add several groups (token 2 token 3 ....) 
thank you

luc-github, 2020-04-19:
what would be the user case?
why need many chat id?

tokaphi, 2020-04-19:
to send notifications to several chat

luc-github, 2020-04-19:
@tokaphi yes that I got it
but why need notification to many chat id ?

tokaphi, 2020-04-19:
to notify people who are in different groups

Ex:  Pierre, paul, jacques 
pierre (chat A, privé)  paul (chat B , public)  jacques( chat A&B)

luc-github, 2020-04-19:
c'est quoi l interet de poster la fin d'un travail d'impression ou de cnc  a plusieurs chat id, il y a deja plusieurs personnes dans le chat id, je suis desole je voudrais comprendre 
pourquoi pierre paul jacque ne sont pas dans le meme chat id si c est pour la meme imprimante, c 'est peut etre clair pour vous mais je ne saisis pas, d'ou ma question

tokaphi, 2020-04-19:
On peut avoir differents groupes de discutions  avec des personnes differentes que l'on ne veut pas forcément avoir ensemble et comme le plugin telegram de octopi permet de le faire je me permettais de demander si il était possible de faire la même chose avec esp3d , une autre question serait-il possible d'envoyer une commande (demande de status par exemple ) via telegram ?

luc-github, 2020-04-19:
pour le multi chanel id il suffirait de mettre   les channels id separes par des points virgules et on process les uns apres les autres mais je ne comprend toujours pas le cas d application.

pour la commande de status je ne connais pas bien l api de telegram mais je pense qu'il faudrait ecrire un bot pour ca, le bot d envois de message ne les lit pas.
donc il faut un bot qui recoit la commande et qui envoit une requete http a esp3d qui repond par une notification

tokaphi, 2020-04-19:
j'avais testé le point virgule entre chanel id mais ça ne fonctionne pas actuellement , il prend seulement le premier.
pour le multi chanel ,ca permet aussi d'avoir par ex exemple un chanel en sourdine et pas l'autre .

luc-github, 2020-04-19:
le point virgule ce n est pas implemente j 'ai dit que cela pourrait etre une solution,  cela eviterait de creer de nouveaux token

luc-github, 2020-04-20:
@tokaphi je vais reposer la question car je voudrais vraiment comprendre et la je ne comprend pas:
Dans quel cas de figure a- t-on besoin de plusieurs chat ID pour une imprimante ?
>On peut avoir differents groupes de discutions avec des personnes differentes que l'on ne veut pas forcément avoir ensemble
 
Pourquoi ? 

>ca permet aussi d'avoir par exemple un channel en sourdine et pas l'autre

Ca veut juste dire qu'on en veut plusieurs mais pas pourquoi, 
Donc pourquoi un channel en sourdine et pas l'autre ?

>to notify people who are in different groups

Je ne dois pas etre habitue a telegram mais quel est le cas de figure ou des gens partageant les meme info sur une imprimante, sont dans differents chat ID, au lieu d'etre dans celui de l'imprimante ?

tokaphi, 2020-04-20:
j'ai par exemple un projet de réalisation d'une CNC où j'imprime pas mal de piéces , j'ai un chanel sur lequel je partage mon projet et veux partager quelques infos de l'impression des differentes pièces, en même temps je voudrais remonter les infos aussi a mon système domotique qui va me permettre par exemple de faire sceenshoot ou éteindre une lumière etc... , dans ce cas je ne veux pas que mon chanel privé de ma domotique apparaisse parmis le chanel de la CNC et pour la sourdine ,  ex le chanel CNC est en sourdine car je ne veux pas etre notifier toutes les 2 minutes par les différents membres en revanche je veux toujours avoir l'info de mon imprimante ..

luc-github, 2020-04-20:
pour le moment il y 2 type de notif 
1 - autonotif : qui envoit l'ip de l'esp au demarrage, dans GRBL_ESP32 et dans MARLIN_ESP32 il peut y avoir plus de notifications 
2- la notification manuelle a mettre dans le slicer ou toute appli qui control l'impression 
`[ESP600]<le message de la notification>`
Contrairement a octoprint ESP3D n'est pas un host sur une carte fille, il envoit des requetes GCODE via le webUI, / telnet a l'imprimante/cnc par serial  pour connaitre le status  ou reagit a des commandes `[ESPXXX]`, dans dans GRBL_ESP32 et dans MARLIN_ESP32 la c'est different car si on imprime depuis la SD il y a plus de possibilitees (c'est implemente dans GRBL_ESP32 mais pas encore dans ESP3DLib pour MARLIN_ESP32)

luc-github, 2020-04-24:
Documentation is done https://github.com/luc-github/ESP3D/wiki/Telegram

commit e525bcb6 (2020-04-14): Add telegram support per @fjriosp request

fixes #439
closed 2020-04-24 by luc-github