luc-github/ESP3D #440: ide compilation error-websocketserver.h

tnap1979, 2020-04-14:
trying to compile with latest IDE, and latest esp3d..
esp8266 with a 16mb eeprom
everything went fine installing and testing flash memory
but trying to compile the esp3d, i get this error

Arduino: 1.8.12 (Windows 10), Board: "Generic ESP8266 Module, 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), no dtr (aka ck), 26 MHz, 40MHz, DIO, 16MB (FS:14MB OTA:~1019KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from sketch\esp3d.cpp:88:0:

syncwebserver.h:25:30: fatal error: WebSocketsServer.h: No such file or directory

 #include <WebSocketsServer.h>

                              ^

compilation terminated.

exit status 1
WebSocketsServer.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

luc-github, 2020-04-14:
did you read installation section of readme for the libraries?

tnap1979, 2020-04-14:
just did that now, it seems to be compiling now
so for anyone with the same issues...
id reccomend to copy the libraries folder from the esp3d-2.1 folder that you extracted from the zip file, into your documents/arduino/libraries folder... i did this and my device is working and esp3d flashed and uploaded

closed 2020-04-14 by tnap1979