luc-github/ESP3D #441: [Question]-compiled, uploaded, but get nothing from device... no wifi ssid shows up...

tnap1979, 2020-04-14:
what could be the issue??
it now uploads, i followed all directions, and everything compiles and uploads, but no hotspot or esp ssid shows up, ive tried to upload and recompile 3 times, and it succeeds but does not do anything afterwards

luc-github, 2020-04-14:
as said you have a big flash, first time  spiff is formated and can take a long time

tnap1979, 2020-04-14:
https://www.amazon.ca/gp/product/B07KF119YB/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

tnap1979, 2020-04-14:
aprox how long do i leave it ??
10min. ? 20?

tnap1979, 2020-04-14:
its been 10min. since arduino ide said soft resetting
the device i have doesnt have a reset button, just a switch to go from uart mode to program mode
and there is still no esp wifi showing up in my phone or my tablet or laptop

tnap1979, 2020-04-14:
there isnt even any flashing light either

tnap1979, 2020-04-14:
so i just flashed the esp8266 blink sketch, and the led is blinking, so this tells me the device is functional...
but why does there not show any esp wifi hotspot...
this is a dilemma i have no idea how to solve

luc-github, 2020-04-14:
enable debug mode and çheck what serial output said on serial monitor

tnap1979, 2020-04-14:
also just flashed with the esp8266 wifi ap sketch
it does show up in wifi lists, in a matter of 2min max...
is there something im doing wrong here??
am i supposed to edit or configure anything in the sketches of esp3d before i upload to the unit ??
ive followed all the directions and it doesnt seem to work for me?
about ready to give up and just bin this thing

tnap1979, 2020-04-14:
ok, ill try the debug mode now

tnap1979, 2020-04-14:
heres the results of the debug

17:55:29.984 -> 
17:55:29.984 -> SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
17:55:34.047 -> mode : softAP(8e:aa:b5:0d:da:f9)
17:55:34.047 -> add if1
17:55:34.047 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
17:55:34.047 -> bcn 100
17:55:34.187 -> bcn 0
17:55:34.187 -> del if1
17:55:34.187 -> usl
17:55:34.187 -> add if1
17:55:34.187 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
17:55:34.187 -> bcn 100
17:55:42.689 -> 
17:55:42.689 ->  ets Jan  8 2013,rst cause:4, boot mode:(1,7)
17:55:42.689 -> 
17:55:42.689 -> wdt reset

luc-github, 2020-04-14:
can you use core 2.5.2 , you are using 2.6.3

tnap1979, 2020-04-14:
It wont upload with 2.5 .2

https://drive.google.com/file/d/1CRySIPrWaEPLniC2nPoNOiYkG4KksvTc/view?usp=drivesdk

luc-github, 2020-04-15:
cannot see the fille

https://drive.google.com/file/d/1CRySIPrWaEPLniC2nPoNOiYkG4KksvTc/view?usp=drivesdk

luc-github, 2020-04-15:
looks like gpio0 is not set to gnd, it works with 2.5.2

tnap1979, 2020-04-15:
How would I fix that up? I have no clue tbh

luc-github, 2020-04-15:
unplug esp board and retry to flash

luc-github, 2020-04-15:
,@tnap1979 any update?

tnap1979, 2020-04-15:
it still fails to upload,
i even tried shorting gpio0 to gnd on the esp-01s

luc-github, 2020-04-15:
how do you flash your esp01?

tnap1979, 2020-04-15:
With a usb stick programmer..
I sent the amazon link to the device I use in a previous comment :-)
It's a programmer that has a switch to go from uart mode to program mode

luc-github, 2020-04-15:
there is no reason you can flash With 2.6.3 and not 2.5 .2, are you able to flash again with 2.6.3?

tnap1979, 2020-04-15:
I'll give it a shot again

nlovrens, 2020-04-16:
Hi. I am having the same issue. Tried on two ESP01 boards. Both with version 2.6.3 and 2.5.2. On both version I have successfully flashed the board but there is an issue with AP signal broadcast. Wifi AP goes up for a brief second and then down again, after that broadcast randomly goes up-down, but there is no time to connect with the PC. I am using inSSIDer for wifi scan and I can see the signal (screenshot attached). 
[image]

I have tired WiFiAccessPoint example on the same board and it works normally.

luc-github, 2020-04-16:
@nlovrens please open a new issue it is hard to help 2 different people on same thread
@tnap1979 has different symptoma than yours as he use a reworked ESP01-> using 128Mb flash and has issue to flash 2.5.2 core
On my side I have no problem with 2.6.3 neither 2.5.2 on reworked ESP01 512KB -> 4MB flash
so need to debug individually to find out your separate problem

nlovrens, 2020-04-16:
ok, thank you. I will open new ticket

luc-github, 2020-04-19:
@tnap1979 any update?

tnap1979, 2020-04-19:
I gave up and moved onto getting klipper installed and running... to much
trouble for this little thing.. lol
Thanks for all the help tho

luc-github, 2020-04-19:
ok happy printing then

closed 2020-04-19 by luc-github